motioneye-project/motioneyeos #699: Flag or Interrupt to RP3 with Jessie

o3aaron, 2017-01-14:
Quick Question.  I got this working perfectly on an RP Zero.  I saw in previous issue discussion MQTT is not yet supported.  Is there another quick way to send a flag/interrupt to another RP3?  It doesnt need to be complex but I just want my RP3 with Jessie to be able to react when motion is detected from my RP zero with motioneye.

sylvaindc, 2017-01-14:
You should be able to run a scipt on motion detection. Which should allow
you to do everything including sending an MQTT message if you have a
suitable client. Alternatively, you could do a simple wget
http://YourPI3/WhateverCall or do something with SSH.

o3aaron, 2017-01-15:
OK that would be perfect.  I thought I saw some there in this forum that mqtt was not supported yet but I will give it a try.  SSH would also work well.  Thank u

sylvaindc, 2017-01-15:
To be honest, i had to google mqtt. If there is some client that you can
interface with from the a script than it should work without integration

closed 2017-06-01 by bortek