motioneye-project/motioneyeos #700: Proble with login

kytcoin, 2017-01-14:
Hi i have the trouble with login vrsion  "motioneyeos-raspberrypi2-20161125"  invalid credetial??
[image]
wht is that?? :-0

kytcoin, 2017-01-15:
ADMIN password not working

johtho, 2017-01-16:
Hi
I have the same problem when i try to login from my iphone, but works when using the anroide tablet.
Have tried to login from  diffrent computers?

kytcoin, 2017-01-18:
i try every time connect from MAC OS X

pescadou999, 2017-01-23:
i just can say no problem for password admin and user with : RPI3 and motioneyeos-raspberrypi3-20160320.img

ccrisan, 2017-01-23:
@pescadou999 does that mean you have login problems with recent versions? If yes, which ones have you tried?

pescadou999, 2017-01-23:
@ccrisan . No sorry i speak French. I have no problem , i think it is a other guy , i have no problem no it is the top with your soft and i make the update direct in the soft and it work goed, great job ccrisan

johtho, 2017-01-23:
Hi, 
yes i cant login with my iphone, but works fine when I'm using anroide tablet.
Installed version 20161125   4.0.1+git37b3595

johtho, 2017-01-23:
Tried using W10 and it also works to login. For me it seems that its only iOS device how have the problem to login.

ccrisan, 2017-01-23:
@johtho try clearing the browser cache on your iOS (if that is even possible).

johtho, 2017-01-24:
Hi, 
I have know cleared the cache.
And I can login again from my Iphone.

Tank you Crisan.
regards /Thomas

closed 2017-01-24 by ccrisan