motioneye-project/motioneyeos #706: samba share

sgiannulli, 2017-01-16:
Hi Calin,
any change to do on some config files apart to activate samba on webUI to let it work?
I'm trying latest pre-release on RPi2 and no shares on my Win7 machine
Thanks

sgiannulli, 2017-01-17:
.......maybe workgroup=MOTIONEY in smb.conf does not help

48296, 2017-02-20:
Hi, I'm using a Raspberry Pi 3 and the options in the gui are on. 
I really need the samba share, how do i activate then 
@sgiannulli : did you get the share working? if yes ,how?

sgiannulli, 2017-02-20:
yes I got it
mount -o remount,rw / to make root partition writable
then changed  workgroup=MOTIONEYE in smb.conf to
workgroup=WORKGROUP

48296, 2017-02-21:
I changed the workgroup. now it shows up in the network but is is giving an error when i try to enter it.

sgiannulli, 2017-02-21:
are you using admin credentials to access the share from windows?
the steps I described above were enough for me on Win7

closed 2017-06-01 by bortek