motioneye-project/motioneyeos #707: Motioneyeos with Xiaofang Wifi Camera

hgelpke, 2017-01-17:
Through hacks here:
https://github.com/samtap/fang-hacks

I have been able to get a RTSP server up and running that works with VLC using this address:
rtsp://IP:554/unicode

When I input this URL into mOS, it says there is an error and recommends a refresh. There is no username or password to the camera, FYI.

ccrisan, 2017-01-17:
Please attach your motioneye.log file here.

tobilap, 2017-01-19:
This camera works fine for me. Did you select TCP/IP when you added the camera?

bortek, 2017-06-01:
Closing this issue due to interest. If still interested please post your question in the forum page.

closed 2017-06-01 by bortek

sushenmathur, 2019-04-15:
@bortek 
the xiaofang cam with fang-hack do not have a username and password for the rtsp stream. Do you leave it blank in the motion eye set up ?

bortek, 2019-04-16:
No idea actually. I don't own that cam and can't test it, sorry. But i suggest testing it. Leave it blank and test it.