motioneye-project/motioneyeos #711: Wrong date with wireless static ip, but correct date via dhcp either wireless or wired.

kondre, 2017-01-18:
I have motioneyeOS installed and everything is working properly. In my router config I have assigned a static IP address to my PI and with a wired connection I get the correct IP and correct date. 
With a wireless connection via (automatic) DHCP I get a random/lease IP address and the date is shown correct.  _Setting current date using http: Wed Jan 18 16:08:39 CET 2017_
When I use manual static IP I get the correct IP address but the date is wrong (1970-01-01), regardless of Date method (http or ntp) - _Setting current date using http: failed_
The same Pi works flawlessly with a standard jessie-os and the date is picked-up correctly via a ntp server.
Any suggestion how to resolve this

bortek, 2017-06-01:
Closing this issue due to interest. If still interested please post your question in the forum page.

closed 2017-06-01 by bortek

solosails, 2017-10-04:
I have the same issue

ccrisan, 2017-10-04:
@solosails it is very likely that your static IP configuration is invalid. In 99% of the cases, the DNS server is wrong. Double check your static IP configuration and make sure your unit actually has full internet access using that configuration (`ping www.google.com` from the motionEyeOS machine should work)

solosails, 2017-10-04:
Hi, ok thanks. I’ll test it out.