motioneye-project/motioneyeos #713: Banana PI Image Broken

tobilap, 2017-01-19:
It seems that the Bananapi Images are broken.
I tested the following Images:
motioneyeos-bananapi-20161212 -> Broken: System is running but not answering Network requests
motioneyeos-bananapi-20161125 -> Broken: System is running but not answering Network requests
motioneyeos-bananapi-20160915 -> Works: Seems to be working fine.

I did not check the other Images.

ccrisan, 2017-01-24:
I can confirm this is happening with recent versions. I can't tell why it is happening though as I haven't changed the kernel or anything related to BPI. It may be the new U-Boot or any of the new BuildRoot packages.

Let's hope it gets fixed with the latest version of BuildRoot. If it won't work, I'll drop support for BPI until someone figures out what's going on.

ccrisan, 2017-03-26:
https://github.com/ccrisan/motioneyeos/releases/tag/20170326 should fix these errors. @tobilap can you try this version and see if the problem persists?

tobilap, 2017-03-26:
ok, ill check. Thanks

tobilap, 2017-03-26:
Works again. Nice Job!

closed 2017-03-26 by tobilap