motioneye-project/motioneyeos #716: Still images almost look inverse, not clear, indistinguishable, incorrect file name

pgbc, 2017-01-20:
motionEye Version 	20161212
Motion Version 	4.0.1+git37b3595
Raspberry Pi 3
When pictures are indistinguishable, the file name is followed by an "m". 
Attached are two files with close time stamps
[image]
[image]

ccrisan, 2017-01-21:
Have you enabled Create Debug Media Files option?

pgbc, 2017-02-20:
I have now, what file should I be looking at for info on my problem?

ccrisan, 2017-02-20:
Make sure the *Create Debug Media Files* us disabled.

pgbc, 2017-02-20:
Okay, I misinterpreted your first reply to thinking that I should turn it on?
So it was off when I first posted my problem of the "inverse" or incomplete images.

ccrisan, 2017-02-20:
Yes, that option should be normally off. The first of your images looks somewhat like the option was on. However the second looks fine to me.

pgbc, 2017-02-20:
The system had worked fine on a B+, but then I got a Pi 3 and moved the camera to it, and that is when my problems started? I just noticed there is a new update out now, so I will try that and see if it makes a difference?

pgbc, 2017-03-01:
Cailin.  I upgraded to 20170212 on the 20th of February, and the problem was solved with the upgrade.
All images look normal since the upgrade.
Thanks for your time.

closed 2017-03-01 by pgbc