motioneye-project/motioneyeos #721: Retrieving the stored videos

Shrirammc55, 2017-01-22:
can someone explain how to retrieve the recorded videos from the sd card? i've tried it pluggin into my desktop, but i cant find the file. I've even tried to copy the file to a usb plugged into the raspberry pi from the terminal in motioneyeos but it doesnt work.

Chiny91, 2017-01-22:
There are several mechanisms and it depends somewhat on how you wish to view the video.

The obvious one is to use the motioneye interface; click on the required streaming picture and various icons appear, hovering over each provides a tooltip about what each one does.   The triangular icon, like a _start_ on a video player, will bring up any videos.

From time-to-time, I use use a GUI sftp client on my computer, having enabled ssh on the R Pi, which works nicely.   You could get the files by command line with ssh, or maybe collect them automatically using an appropriate utility such as the recently provided _rsync_ , although that is perhaps a bit geeky if effective.

Shrirammc55, 2017-01-26:
Thanks for the info Chiny91, i've got the issue solved.

closed 2017-01-26 by Shrirammc55