motioneye-project/motioneyeos #723: ONVIF IP cameras with MotionEye

SuneBielefeldt, 2017-01-22:
Is it possible to integrate IP cameras as follows ONVIF standard into the Motion Eye ecosystem as an external camera ?

ccrisan, 2017-01-23:
As long as your camera has a JPEG (snapshot), MJPEG or RTSP URL, it can work just fine with motionEye(OS).

bortek, 2017-06-01:
Closing this issue due to interest. If still interested please post your question in the forum page.

closed 2017-06-01 by bortek

SuneBielefeldt, 2017-06-01:
It is possible to buy dual stream cameras with ONVIF and mjpeg support. Here is an example of such a camera. I will try it in the near future.

Hikvision® DS-2CD2142FWD-I 
4MP WDR Dome Network Camera with DC12V & PoE
(Waterproof Day Night Motion Detection PoE)
30m IR

http://www.lightinthebox.com/hikvision-4mp-wdr-fixed-dome-network-camera-with-dc12v-poe-waterproof-day-night-motion-detection-poe-dual-stream-remote-access-plug-and-play-30m-ir_p5546865.html
 
[image]
Ü
[image]

waitfor1t, 2017-08-13:
I find more cameras with ONVIF without an RTSP stream (that I can find)... not sure how that happens.. I will try a port scan to see what ports it may be using. Basically I think better ONVIF support would be very helpful.