motioneye-project/motioneyeos #724: Video length 0 secs with extremely high frame rate

akashskypatel, 2017-01-22:
All of the videos generated are 0 secs long and have extremely high frame rate.

Is this an issue with motion dameon?
I have tried searching for similar issues on both motioneye and motion and have not seen anyone reporting this issue. I have also tried this at the lowest settings but i get the same exact results.

I am running this on Raspberry Pi 2 Model B with a NoIR Camera module

[image]

My settings:
[image]

[image]

[image]

cgpgh, 2017-01-23:
I faced a similar issue today. 0 Sec video.

chevyman142000, 2017-01-23:
I believe I am facing a simliar issue as well. I have a brand new setup with a Raspberry Pi 3 and a Foscam camera. Only difference I'm seeing is that I have mine set to record movie continuously as I have a 3tb external hard drive connected. When I download the video so that I can watch it, it downloads a zero kb file that I am unable to play back. When looking at the interface though, it says the file is like 6 or so gb in size.

ccrisan, 2017-01-23:
I can't say anything about the other cases, but @chevyman142000 in your case you're trying to download a relatively large file via motionEye which is unfortunately problematic (things aren't optimized enough there and a lot of CPU is required). Timeouts or incomplete files may be retrieved.

chevyman142000, 2017-01-23:
Thanks for the reply. Is this something you are working on for future updates? What is your recommendation for continuous recording? Right now I have them at 4 hours in length. Maybe do a shorter time so they are smaller files?

ccrisan, 2017-01-23:
@chevyman142000 I'm not currently working on this problem and frankly I don't have any ideas on how to optimize it. I'm sorry about that.

closed 2017-06-01 by bortek