motioneye-project/motioneyeos #725: Can`t add OV5647 camera module to RPI3 motionEye

svizc, 2017-01-23:
Hello,

i am new in motionEye OS.

Few days ago i bought new RPI 3.  Reason of buying it is i would like to run motionEyeOS on it with a few cameras (wifi and cable connected). So with win32diskimager i installed motionEye on sd card and sterted it. Changed admin password, added everything but when it comes to the point when i want to add 
[url=http://www.ebay.com/itm/261582953144]THIS CAMERA[/url] i can`t find it on the list. Where option Local Camera is, under there i can`t chose nothing (it say`s no cameras). 

motionEyeOS is up to date, it`s latest one "motioneyeos-raspberrypi3-20161212.img.gz".

PS. in november 2k16 i was trying this system with another RPI 3(i am ussing that one now for TV) and it worked perfectly. Version was 20160919 and  i could add camera with no problem.

[image]
[image]
[image]

Pictures added and sorry for my english :)

sylvaindc, 2017-01-23:
Are you sure the camera connector is attached correctly?

vikingscania, 2017-01-24:
The cable is attached to wrong connector. Use the other one between hdmi and headphones.

svizc, 2017-01-24:
Hello,

it doesnt work on that one either... just removed IR modules and it also doesnt work...

vikingscania, 2017-01-24:
Hi again,
It could also be that the connector on the backside of the camera is loose.
Happend for me once.

svizc, 2017-01-24:
Ok, will try that at home and report what happened... Thanks

svizc, 2017-01-30:
Its the same, i cant add camera...

sylvaindc, 2017-01-30:
Can you send another picture of your setup

VincentvanDuijn, 2017-05-12:
i am having the same issue

svizc, 2017-05-19:
Did you already figured out where is the problem? I will try one more time now with newest version of the OS and the i will report what happened...

VincentvanDuijn, 2017-05-19:
my fix was 

sudo nano /etc/modules

and place bcm2835-v4l2 keep in mind it is an L at 4l2

closed 2017-06-01 by bortek

svizc, 2018-01-29:
@VincentvanDuijn and on the end this complitely fixed the problem for you, no more problems?. Few day ago i had a little time and once again i tried to add camera module. This time in RPI1 model and RPI3 model, they are both on latest OS 20180101. I added line bcm2835-v4l2 in modules, but this time i get this on both models of Raspberry Pi...
[image]

Any ideas?

svizc, 2018-02-01:
One stupid question, looking all over, how do i edit, actually on the end save file modules file when i put "bcm2835-v4l2" in there. It says that file modules is unwritable, hot to set all the permisions to that file? I am root, but still... tried chmod 777 and nothing?

ccrisan, 2018-02-01:
@svizc https://github.com/ccrisan/motioneyeos/wiki/FAQ#why-do-i-get-read-only-file-system-when-trying-to-edit-a-file

Also, if you want to load a module, see https://github.com/ccrisan/motioneyeos/wiki/modules. However, on a Raspberry Pi, `bcm2835-v4l2` is already present in the modules file. I don't see why you would readd it.

svizc, 2018-02-07:
Still didn`t figure out where is the problem that camera doesn`t work. On both models, RPI 1 and RPI 3 i can`t add camera. I tried, camera works perfectly, but not with motioneye. 

ccrisan can you help me?

ccrisan, 2018-02-08:
Can you try and make a screencast or series of screenshots showing what you exactly do from the very beginning to when the camera doesn't work?

cablegunmaster, 2019-09-23:
I had an similar issue and it all comes down to me to wiggling the cable and rebooting a few times and it will find it.

Basically its a reinsert the cable. make sure its tight.  Reboot the raspberry pi and see if the camera is found kind of operation. 
If not ?
Retry, and recheck both side of the connection even if you are sure. Only after 15 times you could say for sure its not the wire/cable issue. (after six times for me).