motioneye-project/motioneyeos #729: Auto updater to 20161212 not working?

silversword411, 2017-01-23:
I see there's a 20161212 release but software check button reporting up to date?

[image]

Chiny91, 2017-01-23:
I suspect you have not checked _Enable Prereleases_ under _Expert Settings_.    You may not want to do that for less well tested versions, although 20161212 has been in pre-release for a while.

I trust you have blanked out (for your post) your passwords and have not really left them blank for the default users 😱

silversword411, 2017-01-24:
Didn't know it was in prerelease status since it was on the main download page, that's definitely the difficulty!

No passwords, because local LAN access only. Would certainly change if I every put it internet facing :)

closed 2017-01-24 by silversword411