motioneye-project/motioneyeos #732: Setup trust and run additional motion commands

cgpgh, 2017-01-24:
Hello,

I am looking to setup trust between motioneye and my linux NAS server so I can rsync/scp files regularly. And also looking to run motion commands like:
on_movie_end scp %f root@192.168.0.64:/media/3A12B39B12B35A99/Video/
to copy the files over after an even as occurred.

I tried setting up trust many of the filesystems are readonly, I understand this is a purpose built OS. Any help/suggestion is appreciated.

ccrisan, 2017-01-24:
You can add your SSH key files to `/data/etc/ssh_authorized_keys`.

cgpgh, 2017-01-24:
Thanks a lot let me try that.

Regards
Cyril.

cgpgh, 2017-02-02:
If I were to make the motion eye tr

hellyberry, 2017-02-02:
You like risky race conditions (on_movie_end), dont you? :) The files are at risk when cache is not synced to disk yet, but it is read instead so cannot be finished written.

I would recommend a regular niced cronjob using rsync via SSH for this task. Can even be configured to delete all files on source once transferred and verified on destination. 

Having tested several motion actions and for the raspberry pi 2 and 3 devices niced regular cronjobs work best to keep the load low for plannable tasks (video conversion, uploads). Or you can 'mount the pi completely' via SSHFS and access all files mounted locally.

cgpgh, 2017-02-07:
Thanks for your suggestions. Can you share with me the niced cron entry. Ill go with your suggestion.in my current vanila motion i use rsync. From the pi to a nas server. I was wanting to do the same here but could get the pi to be trusted by my nas server.

Chiny91, 2017-03-15:
I would think that you just prefix your crontab entry with the niceness value.   Looking at one of my R Pi (not motionEyeOS), I see I have also added io niceness as well, presumably being concerned at what happens on the R Pi ethernet.

Anyway, you might end up with something like:
`nice -n 19 ionice -c 3 on_movie_end rsync here there`
Tweak to suit; I've not checked on my motionEyeOSs.

closed 2017-06-01 by bortek