motioneye-project/motioneyeos #733: Feature Request - Enable / Disable Motion detection by Mobile connected over WLan?

CampoES, 2017-01-24:
Hi ccrisan,

ok, there are many issues to this wish, but perhaps (I don't know this) this idea is easy progammable (for you) and include an automatic login and logout. Perhaps it is possible for someone to write a script/cronjob that can make this job?

Would it be possible to check by motioneyeos on the WLan-router if a (or two) mobile is logged in? I think most people who use your system have a mobile an use it at home over wlan. So if someone come home the mobile log into the wlan automaticly. The motioneyeos should then deactivate motiondetection. If someone turn off the mobile or leave the house/flat the router will loose the connection to the mobile an motioneyeos should then activate motiondetection.

Michael

CampoES, 2017-01-24:
EDIT:

with this small shell-script I can check if a mobile with fixed IP is logged in:

`#!/bin/bash

if echo $(ping -c 1 192.168.1.xx) | grep Unreachable &>/dev/null;       # Mobile
        then echo "activate Motioneyeos.";
        else echo "deactivate Motioneyeos";
fi`

But how can I toggle motion detection? Is there a way?

CampoES, 2017-01-24:
EDIT 2:

can I do that this way:

`#!/bin/bash

if echo $(ping -c 1 192.168.1.xx) | grep Unreachable &>/dev/null; # Mobile
then motion -c /data/etc/motion.conf;                                             # activate motion
else if test -f /tmp/motion.pid; then kill $(cat /tmp/motion.pid); fi;  # deactivate motion
fi`

snagglewest, 2017-01-27:
https://github.com/ccrisan/motioneyeos/issues/464

I played around for a while but never got it to work as I'm woefully deficient in programing.  If you figure it out please post.  I'd really like to make it work.

CampoES, 2017-01-28:
Hi snagglewest,

there is a solution for "normal" Linux-systems: see issue #203. 

I changed some parts for my motioneyeos on rasp pi3+ like this an it works (comments are in german). The script is called every x minutes over a cronjob:

#!/bin/bash
#########################################################################################################################
#                                                                                                                       #
#       MotioneyeOS durch Anwesenheit/Abwesenheit eines eingeschalteten Mobiltelefon aktivieren/deaktivieren   #
#                                                                                                                       #
#########################################################################################################################
Mobile_IPs=("192.168.x.xx" "192.168.x.xx");                             # Mobile numbers
Mobile_present=0;

for Mobile in "${Mobile_IPs[@]}"
do
        ping -c 1 $Mobile &>/dev/null && Mobile_present=1
done

if test -e /tmp/motioneye.pid;                                          # läuft Motioneye?
then                                                                    # ja, Motioneye läuft bereits
        if [[ $Mobile_present -eq 1 ]];                                 # Mobile at home?
        then                                                            # ja
                /bin/bash /etc/init.d/S85motioneye stop;                # dann Motioneye stoppen
        fi
else                                                                    # nein, Motioneye läuft nicht
        if [[ $Mobile_present -eq 0 ]];                                 # Mobile at home?
        then                                                            # nein
               /bin/bash /etc/init.d/S85motioneye start;                # dann Motioneye starten
        fi
fi

Hope this helps you too.

Michael

nicolasbisi, 2017-03-15:
I did a lot of testing on another system trying to detect if my phone (iOS) is connected or not to the network. In my case, it was always terrible due to the phone entering a sleep mode.
My solution was to use Bluetooth instead. I was actually really impressed on how reliable it is. This would be a nice feature to integrate to motioneye.

bortek, 2017-06-01:
Closing this issue due to interest. If still interested please post your question in the forum page.

closed 2017-06-01 by bortek

sfgabe, 2018-02-12:
If anyone is looking for this, I was able to build on the above (thanks @CampoES ) and turn on motion detection only when I'm not home. I didn't want to turn the entire system off because it has other cameras running through it that I want to keep running. 

First I made an action button containing the following script:

`#!/bin/bash

Mobile_IPs=("YOUR.PHONES.IP.ADDRESS"); # Mobile numbers
Mobile_present=0;

for Mobile in "${Mobile_IPs[@]}"
do
ping -c 1 $Mobile &>/dev/null && Mobile_present=1
done

if test -e /tmp/motioneye.pid;
then # ja, Motioneye läuft bereits

# in case this is a transition turn the motion detection on or off
motion_key="@motion_detection"
config_file="/data/etc/thread-1.conf"

if [[ $Mobile_present -eq 1 ]]; # Mobile at home?
then # yes
    #debug echo "You are at home"
    sed -i 's/@motion_detection on/@motion_detection off/g' $config_file
    ../../sbin/reboot
else # no
    #debug echo "You are away from home"
    sed -i 's/@motion_detection off/@motion_detection on/g' $config_file
    ../../sbin/reboot
fi
fi`

After this is set up, you can trigger it with a web request like:
`http://your.publicport.com:xx/action/1/preset1/?_=YOURGENERATEDKEY&_username=YOURLOGIN&_signature=YOURPASSWORDSIGNATURE`

You can get your url request above by using the Chrome Inspect option and clicking on the action button (It helps to "record" just that action so you can narrow it down) from there you can see the request that is sent.

Initially I had it sending a webhook to IFTTT but the geolocation from my Android device was terrible, so I switch to sending it via Tasker App through the Net -> Post option. 

When I leave it triggers via geolocation, so when I'm out of range of my home network it will check for my phone IP, find it's not there, reset the thread-1.conf and reboot. When I come home it triggers via wifi connecting to my network (since I may enter the geolocation ring before my phone connects to the network), checks for my phone IP, resets the thread-1.conf, and reboots.

If @ccrisan has an idea about how to accept the changed conf file without reboot, that would be the best possibility, but I wasn't able to figure that one out.

milmber, 2018-05-12:
@sfgabe thanks for the script. Variation below of your script which uses arp-scan for devices which can't be pinged. This one is specific to starting and stopping motion, motioneye and homebridge but can be adapted and plugged into cron.

```
#!/bin/bash
  
MAC_ADDRESSES=("you.mobile.wifi.mac.address"); # DEVICE numbers
DEVICE_PRESENT=0;
    
for DEVICE in "${MAC_ADDRESSES[@]}"
do
	OUTPUT=$(sudo arp-scan -l | grep $DEVICE)
	if [ -n "$OUTPUT" ];
	then
		logger "$DEVICE detected on network"
		DEVICE_PRESENT=1;
	else
		logger "$DEVICE not detected on network"
	fi
done

INACTIVE=$(systemctl status motioneye | grep "inactive")
if [ -n "$INACTIVE" ];
then
	MOTIONEYE=0;
else
	ACTIVE=$(systemctl status motioneye | grep "active")
	if [ ! -z "$ACTIVE" ];
	then
		MOTIONEYE=1;
	fi
fi

if [[ $DEVICE_PRESENT -eq 1 ]] && [[ $MOTIONEYE -eq 1 ]]; 
then
	logger -s -t $(basename -s .sh $0) "You are at home & motioneye is active. Stopping motioneye."
	sudo systemctl stop motion motioneye homebridge
elif [[ $DEVICE_PRESENT -eq 1 ]] && [[ $MOTIONEYE -eq 0 ]];
then
	logger -s -t $(basename -s .sh $0) "You are at home & motioneye is already inactive. Nothing to do..."
elif [[ $DEVICE_PRESENT -eq 0 ]] && [[ $MOTIONEYE -eq 1 ]];
then
	logger -s -t $(basename -s .sh $0) "You are away & motioneye is already active. Nothing to do..."
elif [[ $DEVICE_PRESENT -eq 0 ]] && [[ $MOTIONEYE -eq 0 ]]; 
then
	logger -s -t $(basename -s .sh $0) "You are away from home & motioneye is inactive. Starting motioneye."
	sudo systemctl start motion motioneye
        sudo systemctl start homebridge
fi
```

sfgabe, 2018-06-07:
@lagunabravo here's a shot of my tasker settings:
[image]

lagunabravo, 2018-06-11:
Brilliant! Thank you @sfgabe

pereorga, 2018-11-22:
Looks like this may be implemented in the future? (**MAC presence on LAN** is listed in https://github.com/ccrisan/motioneyeos/wiki/Roadmap)

Meanwhile, would it make sense to have page in the wiki with these instructions?

Apathialol, 2019-01-11:
I modified @milmber script to be used on another Ubuntu/Raspbi server because I use MotionEyeOS on one camera and I couldn't figure out how to disable motion by using a stop server command, plus it's on an Rpi B+ so it doesn't do very well with extra tasks.

My script detects if any of the 3 devices are on the network using arp-scan, then checks whether or not motion detection is actually active on my MotionEyeOS cam, then runs through enabling or disabling it accordingly. It also logs everything because I don't know what I'm doing and needed to figure out what was happening as I was modifying this script to actually work properly.

```
#!/bin/bash
MAC_ADDRESSES=("12:ab:c3:45:67:d8" "12:ab:c3:45:67:d8" "12:ab:c3:45:67:d8"); #Add device MAC addresses
DEVICE_PRESENT=0; #Boolean variable for device presence
MOTIONEYE=0; #Boolean variable for motion on or off
LOGFILE="/var/log/detection.log"; #logfile location
DATE=`date "+%Y-%m-%d %H:%M:%S"`; #get current date

for DEVICE in "${MAC_ADDRESSES[@]}" #For each device in MAC_ADDRESS array
do
	OUTPUT=$(sudo arp-scan -l | grep $DEVICE) #Find device using arp-scan
	if [ -n "$OUTPUT" ]; 
	then
		echo "$DATE: $DEVICE detected!" >> $LOGFILE #Print result to log file
		DEVICE_PRESENT=1; #Device is detected so change boolean to on
	else
		echo "$DATE: $DEVICE not detected!" >> $LOGFILE #Print results to log file
	fi
done

if curl -sSf "http://192.168.1.100:7999/1/config/list" | grep -q "emulate_motion</a> = on" #Connect to webserver and check if motion detection is on
then
	echo "$DATE: Motion is on!" >> $LOGFILE #Write to log file
	MOTIONEYE=1; #Motion is on so change boolean to on
else
	echo "$DATE: Motion is off!" >> $LOGFILE #Write to log file #Write to log file
fi

if [[ $DEVICE_PRESENT -eq 1 ]] && [[ $MOTIONEYE -eq 1 ]]; #If you're home and motion detection is on
then
	echo "$DATE: You are at home & motioneye is active. Stopping motioneye." >> $LOGFILE #Write to log file 
	wget -q -O- "http://192.168.1.100:7999/1/config/set?emulate_motion=off" >/dev/null #Connect to URL to turn off motion detection

elif [[ $DEVICE_PRESENT -eq 1 ]] && [[ $MOTIONEYE -eq 0 ]]; #If you're home and motion detection  is off
then
	echo "$DATE: You are at home & motioneye is already inactive. Nothing to do..." >> $LOGFILE #Write to log file
elif [[ $DEVICE_PRESENT -eq 0 ]] && [[ $MOTIONEYE -eq 1 ]]; #If you're not home and motion detection is on
then
	echo "$DATE: You are away & motioneye is already active. Nothing to do..." >> $LOGFILE #Write to log file
elif [[ $DEVICE_PRESENT -eq 0 ]] && [[ $MOTIONEYE -eq 0 ]]; #If you're not home and motion detection is off
then
	echo "$DATE: You are away from home & motioneye is inactive. Starting motioneye." >> $LOGFILE #Write to log file
	wget -q -O- "http://192.168.1.100:7999/1/config/set?emulate_motion=on" >/dev/null #Connect to URL to turn motion detection on
fi

```

After you create the script you need to make it executable (sudo chmod +x script.name), and then add it to crontab (sudo crontab -e, * * * * * /path/to/script.name). I have mine running every minute and it's pretty reliable if I use multiple devices (phones usually disconnect from the network after a while so it motion would randomly start up after not using my phone for a while so I have it detecting my phone and laptop (wireless and ethernet) since I'm usually on one or the other when I'm home)

I'm 100% the script can be improved but it works well enough for me.