motioneye-project/motioneyeos #740: wpa_supplicant: failed (8192cu)

dpradaro, 2017-01-29:
Hi there. I have been a user of motioned OS for a couple of weeks on a RPi3 and I love it! Now, trying to use it on a RPi0 has been a huge pain in the neck. Initially, I was utilizing the wrong WIFI dongle and only realized it once I installed Raspbian on the Pi and tried using it. Went and purchased a new dongle (Edimax n150) which works just fine in Raspbian but will not connect to motioneye OS no matter what I do. I have read every post and done every recommendation but nothing works.

Ultimately I am getting a wpa_supplicant: failed (8192cu) message. What does that mean and how can I fix it? Please HELP!
[image]

strofse, 2017-02-02:
I have this problem too.

sylvaindc, 2017-02-02:
Just thinking out loud, but have you tried adding a kernel mod for the
specific network chip?

I seem to remember the wiki saying something about that,  but I'm not sure

ccrisan, 2017-02-06:
@dpradaro does this happen with various versions of motionEyeOS or is it just the latest?

closed 2017-06-01 by bortek