motioneye-project/motioneyeos #742: Feature Request - Audio

lawrencecrosier, 2017-01-30:
Would it be possible to add the capability to select, and listen to, an audio source from the web interface?  It would be nice to listen in on either a microphone embedded in a camera, such as a cheap Play Station Eye, or from a stand alone microphone.  Thanks for your help.

bortek, 2017-02-14:
I backup this request. I would like to have an audio detection mechanism and play in sound for X seconds. There are USB microphones that work for Pi like this one
http://www.ebay.com/itm/232080637075?_trksid=p2057872.m2749.l2649&ssPageName=STRK%3AMEBIDX%3AIT

danpowpow, 2017-02-20:
+1 for audio detection!

Isaac96, 2017-02-20:
+1-just started using this on Pi 3 with usb cam.

bortek, 2017-02-20:
@Isaac96 how exactly are you using , have you implemented your own code for audio?

Isaac96, 2017-02-20:
I am using motionEyeOS, no audio.

gbfromhb, 2017-04-15:
+1 for me. Great idea for many reasons. Just to play triggered sound for deterance, or two-way would-be amazing for the front door.

joshcam, 2017-05-28:
+1 for audio streaming support using any ALSA supported usb mic.

fredericcollin, 2017-06-23:
+1 for audio streaming support! All USB webcams have integrated mics, it's a pitty not to use it!

mja001, 2017-06-23:
+1 for audio streaming. I'd love to use my Logitech C920 HD USB Webcam for this - it does hardware H264 compression and has stereo mics for audio.

jasaw, 2017-08-24:
What is this feature request about?
Audio detection i.e. trigger an event and start recording when some sound is detected?
OR
Add audio channel in recording & streaming i.e. issue #105 ?

lawrencecrosier, 2017-08-31:
Jasaw, thanks for bringing the #105 request to my attention.  It was what was looking for initially, but triggering recording on a loud sound would also be quite useful.

timz3818, 2018-01-25:
+1 on this being useful. I'm currently using motioneyeos with a pi camera as a wildlife camera trap and it would be great if I could hook up a USB mic and have audio recorded as well.

1dren, 2020-02-05:
i want this issue

89636443592, 2020-05-30:
+1 for Audio/Sound Support
Let's make Great Software Greater!