motioneye-project/motioneyeos #743: Raspberry Pi zero v1.3 Issues solved including reboot with Wifi.

ItsJoeEnterprises, 2017-01-31:
Hello all, My name is Joe and I sat and had a hard time trying to figure out how to get motion eye OS to and couldn't find a fix for it. SO with the raspberry Pi Zero you have to use the Raspberry Pi image (Not the Pi2 or Pi3). 

Imaging the device is easy, with windiskimager and you can youtube it for an easy tutorial. If you are not going to use the ethernet cable you need to set up wifi before you boot. I tried other troubleshooting options (Write the wpa_supplicant.conf file in notepad and saving it which didn't work). I found the easiest way is to use Rasbian (Another raspberry operating system) to write the program. I found the only way to do this is through root terminal (It has to be Root terminal which you can access by going into preferences menus and turning on root terminal in the menu. You will get a warning that states you are entering root terminal which means you can change anything. Next you type in "sudo nano /etc/wpa_supplicant/wpa_supplicant.conf" this will open the wpa supplicant file to change, you will find nothing inside of this and you will need to write:
network={
              ssid="yournetworkname"
              psk="yourpassword"
}
exactly like that. At this time I plugged a usb drive into the PI and when a pop up asks how to want to view files on usb drive choose the file manager and copy the url to the USB drive (Should look like this or similar pi/"a bunch of randon letters and numbers"). Next go back to root terminal and hit control X to exit and it will ask you to save and you hit Y and it will ask you to save it. It will say etc/wpa_supplicant/wpa-supplicant.conf and you will want to erase that and paste the url for the usb drive you copied and save it there. It may stay open and show nothing changed but don't worry all changes are on the file in the usb drive. Next eject the usb drive from the Pi and go to your computer plug it in and copy the file to the sd card (drag and drop from usb to SD card). 

Now you can do the first boot and it should ignore the ethernet and boot up with wifi connection. Of course there is probably other ways to do this and you need a windows computer, a usb, a usb hub if you are using a rasberry pi zero (I have a pi zero and a pi 3 so I did it on pi3) and a second micro SD card for rasbian. But this worked for me and if you have any questions feel free to ask.

SuperEvenSteven, 2017-01-31:
Gotta say setting wifi up on a Raspberry A+ 512mb has been a waste of 1 hour so far. I've tried the writeimage.sh on ubuntu with a fresh sd card which wrote the 2 partitions but failed to mount /dev/sdc1 and exited, so I read the script and manually created the wpa_supplicant.conf file myself. 

At first boot it expanded the filesystem and copied necessary files (including wpa_supplicant.conf). No joy. I get a **wpa_supplicant failed (8192cu)** message. 

Instead I tried writing the image from windows using windisk imager 32 instead and placed the wpa_supplicant.conf in the boot partition. Same error. 

I've followed all the troubleshooting regarding Pi A+ owners and general wifi config guidelines to ensure correct ssid and psk spelling, checked wifi channel is within range and is WPA2. Why isn't there an/etc/network/interfaces and why can't we emergency boot into the terminal instead of this continual boot loop? 

Mounting/unmount the sdcard is a pain just to make a config change.

SuperEvenSteven, 2017-01-31:
If this helps anyone make sure you surround your ssid and preshare key in double quotes in case you need to escape any characters. 

In my case had a period in my ssid and an exclamation mark in my psk. i.e.

update_config=1
ctrl_interface=/var/run/wpa_supplicant
network={
    scan_ssid=1
    ssid="some.SSID"
    psk="somPassword!"
}

ccrisan, 2017-01-31:
@SuperEvenSteven the reason why there is no `/etc/network/interfaces` is that motionEyeOS is *NOT* a general purpose Linux OS. Thanks to the simplification of the boot mechanism, it boots considerably faster than, say, Raspbian.

You should see motionEyeOS as a firmware of a surveillance camera: you don't tweak too many network-related parameters on your netcam and you don't have emergency boot on your netcam.

Instead, whenever your netcam detects a problem (i.e. network connectivity or alike) it tries to recover by quickly rebooting, which is more likely to fix the problem than doing nothing - hence the reboot loop. It's not perfect but it's the best it can do in most of the cases.

ItsJoeEnterprises, 2017-01-31:
I left out 
update_config=1
ctrl_interface=/var/run/wpa_supplicant
Never added it And I found I had to create wpa_supplicant in rasbian instead or it wouldn't work. 
I personally enjoy the tinkering and problem solving that's why I have a pi.

willbird, 2017-01-31:
The easiest way I found to do Pi Zero is just a device like this.

http://a.co/00qukUH

Plugable USB 2.0 OTG Micro-B to 10/100 Fast Ethernet Adapter for Windows Tablets & Raspberry Pi Zero.

Makes it super easy....for me anyway :-).

ItsJoeEnterprises, 2017-01-31:
That only works if you don't want to use Wi-Fi.

willbird, 2017-02-02:
I use it to set the WIFI SSID and password :-).....easy peasey.

Plug the Zero into network, find the IP, login, setup wifi...done.

closed 2017-06-01 by bortek