motioneye-project/motioneyeos #744: Starting wpa_supplicant: failed (8192cu)

strofse, 2017-02-01:
Cant boot a pi zero with the Edimax Ew-7811Un wifi dongle and get the "Starting wpa_supplicant: failed (8192cu)" in the boot.log and the pi zero goes in to a boot loop.

Tried advice in this thread: https://github.com/ccrisan/motioneyeos/issues/19

Disable modprobe works but is there something that can be done without disabling modprode? The rtw_ips_mode is not a parameter in my modproble conf (using version motioneye 20161212) on a pi zero v1.3

Thanks for any help!

strofse, 2017-02-02:
I see now that there there is no support for 8192cu in /sys/module/ in the A/A+/ and Zero version om motioneyeos, but there is support in the raspberry pi 2/3 version (where the dongle works fine).

Is there a reason for not supporting 8192cu in this version? It would be great if you can add support to this otherwise great os! I would really like to use the Raspberry pi zero with motioneyeos

ccrisan, 2017-02-02:
It's the exact same kernel for all versions. What happens if you run `modprobe 8192cu`?

strofse, 2017-02-02:
Nothing happens, just returns to command prompt

strofse, 2017-02-02:
Ah sorry, my mistake, it is there, but still nothing happens with modprobe

strofse, 2017-02-02:
I should add that I have tried to troubleshooting wifi in the wiki section, and it is not related to said/psk or power supply strength or anything like that. I can use the dongle on motioneyeos with a raspberry pi 2, but not on the pi zero.

coproc79, 2017-02-06:
I'm running into the same error with a Pi2 and using a WiFi dongle.
 modprobe 8192cu  runs, but nothing happens.
I have the correct WPA2 PSK key.

I have used this dongle with the Pi2 with a normal Rasbian build.

peternsomerset, 2017-02-08:
Sorry to raise this here but I am having the same problems with the latest release and both my realtek dongles (Edimax EW-7811Un and Airlink AWLL6075).  They both work on latest Raspian,  they are detected OK and the appropriate drivers are loaded but do not run with either GUI input of ssid and psk or manually inserting them in /boot/wps_supplicant.conf,  and the GUI states there is no wireless interface.
I have tried this for hours and have made a new image but still does not work.
Is this me or the software!? 
I do have a 24 length of password with 4 numbers,  is that a possible problem? 

Thanks
Peter H

ccrisan, 2017-02-08:
@peternsomerset 

1. What motionEyeOS version ("the latest" is not a valid answer!)?
2. Don't try editing `/boot/wpa_supplicant.conf` (I assume you meant `wpa_supplicant.conf` and not `wps_supplicant.conf`) *after* the first boot, because it won't be taken into account.
3. A 24-characters password should have no negative impact. Nor should numbers in it.

peternsomerset, 2017-02-08:
Hi,

On 08/02/17 20:34, Calin Crisan wrote:
> @peternsomerset
>
> 1. What motionEyeOS version ("the latest" is not a valid answer!)?
motionEye Version 	20161212
Motion Version 	4.0.1+git37b3595

> 2. Don't try editing `/boot/wpa_supplicant.conf`
OK, I had read to do that on a site.
  (I assume you meant `wpa_supplicant.conf` and not `wps_supplicant.conf`

Yep -was a typo!

) *after* the first boot, because it won't be taken into account.

OK so the script onlt works for initial boot.

> 3. A 24-characters password should have no negative impact. Nor should numbers in it.

OK

I have a Pi 2 and looked in your "tweaks"pages and assumed the wifi 
script looked for wpa_supplicant.conf in /boot then put it /data/etc.

Just looked in /data/etc/ and the file is there but not configured for 
my network so I assume that's the problem.
I thought I had tried putting it in /data/etc/ before and it did not 
work, but I could have made a typo with my long password!

I had also read in your troubleshooting page (1Jun 2016 1 revision) to 
put it in boot partition of the card, which I assumed was /boot.

So the question is,  where do I put the .conf file with my network 
details and do I have to delete any other wpa_supplicant.conf files 
before I write the file?

Thanks
>

-- 
Peter H

ccrisan, 2017-02-09:
You can place your wpa_supplicant.conf file on /boot but *before* the very first boot (i.e. while the SD card is plugged into the machine you use to write the OS image).

If you want to edit it afterwards, while your motionEyeOS is running, just edit the `/data/etc/wpa_supplicant.conf` file.

peternsomerset, 2017-02-09:
I have copied this into /data/etc/wpa_supplicant.conf file:
update_config=1
ctrl_interface=/var/run/wpa_supplicant

             network={
             scan_ssid=1
             ssid="my ssid"
             psk="my password"
}

It still doesn't work
The output loops as it cannot find wireless or wired (cable removed).
It gets to:
root password done
then
Starting wpa supplicant
(waits 20 secs)
starting wps supplicant: failed (9192cu)
Configuring wireless network: no link
Configuring wired network:
(waits 10 sec)
and reports no link 
Ethernet no connection available rebooting

Have I got the correct wpa_supplicant.conf settings or is there something else causing it no to work on both dongles?
(this was using the Edimax)

Thanks

bender-rulez, 2017-02-09:
I got the same issue... Testet with Raspberry Pi 1 also tested now on Raspi 3 Mobel b... WIFI also not working with onboard WIFI....

strofse, 2017-02-09:
My problem exactly

bender-rulez, 2017-02-09:
I tested every version down to motioneyeos-raspberrypi3-20161027.img.gz
None is working with preconfigured wifi networksettings (command line option by writing the image to the card).... I assume this was never tested :-)

ccrisan, 2017-02-09:
@bender-rulez I always use the preconfigured wifi settings all the time and it works fine here.

When you say you tested every version, do you mean various releases or the same release on various boards? Anyways, thanks for taking the time to run these tests.

Could you please elaborate a bit what you did to run a test? Answers to the following points would help me:
 1. Did you use the `writeimage.sh` script with the `-n ssid:psk` option? Or perhaps you manually prepare a `wpa_supplicant.conf` file on your freshly written boot partition?
 2. After you try and unsuccessfully configure your wifi the way you described, could you please somehow recover your `/data/etc/wpa_supplicant.conf` file and attach it here, **after removing sensitive data**?
 3. Does your boot log say something like "Starting wpa supplicant: failed (...)"? Or do wifi error messages vary from case to case?
 4. Did wifi ever work for you with motionEyeOS at all? If yes, how did you configure it?
 5. Have you checked [this wifi troubleshooting article](https://github.com/ccrisan/motioneyeos/wiki/Troubleshooting#wireless-connection-problems)?
 6. Does it work with Raspbian?

bender-rulez, 2017-02-09:
Hi, I just tested 20161212 pre-release again, after switching my WIFI from channel 13 to 11... 
But I had to edit wpa_supplicant.conf in /data/etc/ manually because it was empty....
Now it works :-)
Raspbian is able to use channel 13

ccrisan, 2017-02-09:
@bender-rulez could you please set the `os_country` in `/data/etc/os.conf` to `JP` or `UK`, or maybe your own country and see if it works?

peternsomerset, 2017-02-09:
Mine was set to US and changed it to UK

Now works fine
Thanks

bender-rulez, 2017-02-09:
set mine to DE, works

ccrisan, 2017-02-09:
I mistyped `UK`. Please try `GB` instead, as `UK` (which doesn't actually exist) will make it default to `00` which is a generic country and I have no idea how permissive is the generic country.

peternsomerset, 2017-02-09:
Well it was permissive enough for me!!
Just changed to GB and it still works.

Should the country have got set automagically of was there something I missed!?

peternsomerset, 2017-02-09:
or course there was a typo!

ccrisan, 2017-02-09:
It appears that channels 13 and 14 (in 2.4 GHz) are prohibited in the US, which motionEyeOS defaulted to. https://github.com/ccrisan/motioneyeos/commit/a459d1b04f0f400d33e8f37900bb970fdbdaf556 sets this to GB (also the default in Raspbian, it seems) which will (hopefully) allow channels 13 and 14 to work out-of-the-box.

closed 2017-02-09 by ccrisan

peternsomerset, 2017-02-09:
I just checked and I had in fact set my router manually to Channel 13 and not left it at Auto because of other WiFi's locally.  
It would not have been quite as obvious if my router kept changing channels!!

Thanks again

MisterTwee, 2022-01-07:
Bit old issue, but I had the same error running motioneyeos on an old rpi 1 using a wifi dongle:
Starting wpa_supplicant failed (8192cu)
It worked before, but I upgraded my wifi router and changed the ssid and password so it failed probably because of the channel 13 or higher issue.
I edited  /data/etc/wpa_supplicant.conf
Changed country=US to country=NL
Now Motion Eye is working perfectly.
Thanks