motioneye-project/motioneyeos #745: GPU et motion detection

philobbe, 2017-02-01:
Bonjour
Félicitations pour cet os extraordinairement bien fait. Et merci à vous de le partager en open source. La qualité de votre travail est remarquable !
Je me permets de vous écrire en français car j'ai vu une réponse de vous en français... peut-être êtes-vous francophone... ce qui serait un honneur... ;-)

J'ai motioneyeos installé sur une raspberry pi 3 avec la picam 8M.
J'ai des questions à propos du paramètre GPU memory. 
Quelle est la mémoire à indiquer pour un Rpi 3 ? Quel est le maximum possible sans avoir de problèmes ?
A quoi sert cette mémoire ? Quelle influence sur le fonctionnement si j'augmente la mémoire ?

Je n'ai pas bien compris non plus la différence entre Frame Change Threshold et Minimum Motion Frames
Pourriez-vous me donner une explication s'il vous plait, je ne comprends pas la différence... ?

Une question plus générale. L'amélioration de votre os est elle encore possible ou est-elle limitée au hardware actuel du pi ? Peut-il par exemple y avoir une amélioration du framerate et de la qualité pour un pi3 en oméliorant l'os ?

Merci d'avance pour vos réponses

philobbe, 2017-02-17:
I translate my previous message because I read in a post that you prefer in English so that everyone benefits.

First congratulations on this extraordinary os.

I have motioneyeos installed on a raspberry pi 3 with picam 8M.

I have questions about the GPU memory setting.
What is the memory to indicate for a Rpi 3? What is the maximum possible without problems?
What is the use of this memory? What influence on functioning if I increase memory?

I did not quite understand the difference between Frame Change Threshold and Minimum Motion Frames
Could you give me an explanation please, I do not understand the difference ...? Both seem to have an influence on triggering.

A more general question. Is your OS enhancement still possible or is it limited to the current hardware? Can there for example be an improvement in framerate and quality for a pi3 by OS enhancing?

Thank you in advance for your answers

ccrisan, 2017-02-17:
The GPU memory doesn't really affect the overall performance, as long as it's at least 128MB.

For *Frame Change Threshold* see [threshold](http://htmlpreview.github.io/?https://github.com/Motion-Project/motion/blob/master/motion_guide.html#threshold), but keep in mind that motionEye transforms it in percents (relative to the image size).

For *Minimum Motion Frames* see [minimum_motion_frames](http://htmlpreview.github.io/?https://github.com/Motion-Project/motion/blob/master/motion_guide.html#minimum_motion_frames).

I don't understand the last question. Sorry.

philobbe, 2017-02-20:
Thank you.

Since you left the possibility to modify the GPU parameter I thought we could optimize according to the model of raspberry.
But then what is the best value for a raspberry pi 3 ? Would anyone have tested?

I will try to explain myself better for the last question but not being an expert in computing excuse me in advance if my question is incoherent. I'll try to explain.
For a raspberry pi 3 for example, will it be possible in the future by improving motioneyeos to have a recording with framerate of 60fps or is this improvement limited by the hardware of raspberry pi 3 or even by the camera?

I do not want to abuse your time. So I will not be offended if you do not find time to answer this last question. Maybe a member of this forum will take the time to inform me too ...
I thank you again for this amazing work and to have the kindness to animate this forum.

ccrisan, 2017-02-20:
The theoretical limit of 60fps is not reachable using the motion(Eye(OS)) setup. For reference, there's motion, then there's motionEye and then there's the OS accommodating them, called motionEyeOS.

When directly accessing the stream from a Raspberry PI CSI camera module using programs such as raspivid, you could reach framerates as high as 60fps, in theory.

However, if you pass the stream through motion, which does a fair amount of processing using CPU, the actual achievable framerate drops significantly.

Now if you want to see that in a friendly web UI (read: motionEye), the stream needs to be passed from motion to motionEye and you lose another (not so significant) amount of achievable framerate.

In all this "pipeline", the GPU memory has little (if at all) effect. The optimal value for reasonable resolutions (1280x720 is what I'd call reasonable) is 128MB. Going above that would just decrease your main OS available memory.

closed 2017-06-01 by bortek