motioneye-project/motioneyeos #747: Crash after changing timezone and restart

Godward-Matjaz, 2017-02-02:
Hi, I have problem, when changing timezone in gui. If I set it to my timezone euro/Ljubljana and I restart my pi zero, it never boots again, it just keeps rebooting... then I have to write new image to my sd and everything works again. I can save all other settings, just not timezone.

Im using pi zero and 2016 12 12 version of motion eye os.

bortek, 2017-02-14:
I cannot reproduce this. I changed TZ to yours and back and all went fine. My setup is pi zero and the latest (as of today) Motioneye. Try reinstalling Motioneye , it might help get rid of the problem.

Request to close the Issue.

Godward-Matjaz, 2017-02-14:
I made fresh install abot 3 times. I can make all the settings. But everytime I change timezone, I can not get motion eye up.

Godward-Matjaz, 2017-02-19:
I upgraded to latest version 2017 02 12 and it works now. Thanks

bortek, 2017-02-20:
Haha told ya ;) @ccrisan Please close this issue. You need to do little clean up, there a way too many open issues. I can contribute if you can give me proper role for the project.

ccrisan, 2017-02-20:
@bortek I could really use some help, especially when it comes to open issues :)

closed 2017-02-20 by ccrisan