motioneye-project/motioneyeos #752: Feature: storage client rather than server

george-vieira, 2017-02-06:
I see there's ftp/smb server and I could potentially run motionEye on the central server but it could be good if the rasp pis could upload via nfs mount or some other network method avoiding storing on sd card or having additional storage added locally (ie big temp buffer USB stick, or disk, etc).

I have a linux server with raid etc, so makes sense to just push to upstream if possible, but I'll might try installing a central server instance, just preferred not to.

hellyberry, 2017-05-30:
SMB or NFS is not the best from a security-appliance perspective.

If NFS would suit your use-case, why not use SSHFS (or alternatively SFTP/SCP)?

george-vieira, 2017-05-31:
I'm fine with anything. You could let the OS take care of the security perspective as anyone can simply setup IPSEC or openvpn so there would be no need to secure at the application layer.
I'm just trying to avoid having the central server having to rsync from remote multiple motioneye instances whereas if they all mount or remote send images to a central system would be better.

closed 2017-06-01 by bortek