motioneye-project/motioneyeos #754: IR Light.py

hud1nk0, 2017-02-07:
Hi. I am a bit new to raspberry. 
I managed to get working my ir light in raspbian, and i woul like to add python script that will boot with motion os. 

`import RPi.GPIO as GPIO, time

GPIO.setmode(GPIO.BCM)
GPIO.setup(18,GPIO.OUT)
def RCtime (PiPin):
  measurement = 0
  # Discharge capacitor
  GPIO.setup(17, GPIO.OUT)
  GPIO.output(17, GPIO.LOW)
  time.sleep(0.1)

  GPIO.setup(17, GPIO.IN)
  while (GPIO.input(17) == GPIO.LOW):
    measurement += 1

  return measurement

# Main program loop
while True:
  if RCtime(17) > 1300:
     GPIO.output(18,GPIO.HIGH)
     print ('LED on')
     print RCtime(17)
     time.sleep(5)
  else:
     print RCtime(17)
     print ('LED off')
     GPIO.output(18,GPIO.LOW)
     time.sleep(1)
`
[image]

george-vieira, 2017-02-08:
Nice work, and a good idea. I'd like to extend that feature to work with the Working Schedule. So it doesn't need to keep LEDS on when the camera isnt setup to record. Ideally you'd put photo sensors I guess. Then have hooks so when the schedule triggers, it trigger external code which turns on or off as per hook details.

hud1nk0, 2017-02-09:
i just dont know, how to make this script start on boot.

george-vieira, 2017-02-10:
Have you tried making a start up service? There;s no rc.local file but you could make your own startup service

Just make sure to make it a function and then fork that function to the background
clone one of the existing ones.
eg. start up services below...
[root@picam1 ~]# ls -l /etc/init.d/S*
-rwxr-xr-x    1 root     root          1228 May 22  2016 /etc/init.d/S00datapart*
-rwxr-xr-x    1 root     root           834 Sep 13 23:11 /etc/init.d/S01mountall*
-rwxr-xr-x    1 root     root           659 Nov 29  2015 /etc/init.d/S02modules*
-rwxr-xr-x    1 root     root           879 Apr 10  2016 /etc/init.d/S03hostname*
-rwxr-xr-x    1 root     root           581 Feb 22  2016 /etc/init.d/S04syslog*
etc.
etc.

hud1nk0, 2017-02-13:
Thanks a lott. that help me.
i just edit S85motioeye file, and on the bottom of the start section add line

> python /usr/sbin/led.py &

closed 2017-02-13 by hud1nk0