motioneye-project/motioneyeos #755: No more network camera with motionEye Version 20161125?

Narno38, 2017-02-07:
Hello,
I have just performed an update from version back in Sept (201609xx) to the latest one. When doing so, my Vivotek cameras were not functional anymore and I had to remove them to be able to try & recreate. However when trying to do so I found out that the two options available now are Remote motioneye camera and Simple MJPEG camera. Cannot setup a network camera anymore.
What could I possibly do wrong?
thanks much !

ccrisan, 2017-02-07:
Make sure you haven't set a custom Motion Binary in Expert Settings.

Narno38, 2017-02-07:
Hi again, motion binary is set to /usr/bin/motion-mrdave: I don't think I've changed it. I'll try & reinstall from scratch. However strange it really happened from the update. Will let you know.

ccrisan, 2017-02-07:
Please clear it out and leave the input empty.

closed 2017-06-01 by bortek