motioneye-project/motioneyeos #756: Starting wpa_supplicant: fsiled (8192cu)

UltraBob, 2017-02-08:
#415 may possibly be the same issue as mine, all the symptoms are the same, but if I have an ethernet cable plugged in, and log into the raspberry pi I can get the wifi connection going from the command line, so it doesn't seem like lack of power from the USB bus is the issue.

I have a Planex GW-USNAN02A which uses the 8192cu chipset.

If I ssh into the rasberry pi and run:

`ifconfig wlan0 up
iwconfig wlan0 essid <ESSID>
iwconfig wlan0 key s:<password_wireless>
dhclient wlan0`

I get connected and get an IP address.

Indeed if I reboot with an ethernet cable attached, ssh in and run ifconfig wlan0 up, it connects to the wifi network I connected it to before.

the wpa_supplicant.log shows me the 

> ioctl[SIOCSIWAP]: Operation not permitted

error shown in #415 

How can I translate the iwconfig commands used above into wpa_supplicant, because as soon as I unplug the ethernet and restart, I go into an endless no network connection reboot cycle again.  I'm using another raspberry pi to configure motioneyeos wifi for a raspberry pi zero, so I'll have no ethernet option then.

UltraBob, 2017-02-08:
Is it possible that I need to set the regulatory domain since I'm in Japan?  I'll go look at that.

UltraBob, 2017-02-08:
That made no difference, I wondered because of the messages in dmesg.log
[dmesg.txt](https://github.com/ccrisan/motioneyeos/files/760248/dmesg.txt)

UltraBob, 2017-02-08:
Let me get all the most recent log messages
[dmesg.txt](https://github.com/ccrisan/motioneyeos/files/760260/dmesg.txt)
[boot.txt](https://github.com/ccrisan/motioneyeos/files/760259/boot.txt)
[messages.txt](https://github.com/ccrisan/motioneyeos/files/760262/messages.txt)
[motioneye.txt](https://github.com/ccrisan/motioneyeos/files/760261/motioneye.txt)

and wpa_supplicant log is simply:

> Successfully initialized wpa_supplicant
> ioctl[SIOCSIWAP]: Operation not permitted

UltraBob, 2017-02-08:
I've been updating the wpa_supplicant config on my mac with a text editor, but I just found a /data/etc/wpa_supplicant.conf which has none of my later changes, so maybe I've just been spinning my wheels?

UltraBob, 2017-02-08:
OK, that fixed it.  I don't know exactly what I did that fixed it since for a good while I was writing a wpa_supplicant.conf file that wasn't being used.  So the first lesson, is after the first boot, ssh into your raspberry pi and edit /data/etc/wpa_supplicant.conf don't trust writing it into the SD card from your computer after the first try.  Here is my config:

```
update_config=1
ctrl_interface=DIR=/var/run/wpa_supplicant
country=JP

network={
    scan_ssid=1
    ssid="ssid_here"
    psk="password_here"
}

network={
    ssid="ssid2_here"
    psk="password2_here"
    key_mgmt=WPA-PSK
}

network={
  ssid="ssid3_here"
  psk="password3_here"
  key_mgmt=WPA-PSK
}
```

closed 2017-02-08 by UltraBob

ccrisan, 2017-02-08:
For the record, here's a post that may help with understanding the various `wpa_supplicant.conf` locations: https://github.com/ccrisan/motioneyeos/wiki/Tweaks#manually-editing-network-configuration.