motioneye-project/motioneyeos #757: How to control screen-flare?

Thomas-O, 2017-02-08:
Please forgive me if this is a stupid question: how can I prevent the image being obscured by frequent over-exposure?

My camera (HP webcam)  is on the ground floor; where 50-60% of the image is of grey overcast winter sky. Any slight changes in the sky creates a screen flare that covers the entire screen. It makes the system all but unusable. Though only during daylight: the image after nightfall is rock-solid.

Thanks for your help.

Thomas

sylvaindc, 2017-02-08:
Can you make a sun-hood?

Thomas-O, 2017-02-09:
Thanks for the suggestion but I'm not sure it'll be much help. I live in Northern England: there isn't any sun. At least not in winter. Just a sky covered entirely by low, grey, sullen clouds.

And anyway,  the camera is embedded within an ornamental cat. I suppose I could try a big cloth cap on its head, but it'll look a little daft and I don't think the missus will approve.

closed 2017-06-01 by bortek