motioneye-project/motioneyeos #765: Alternative to working schedule

ant-thomas, 2017-02-12:
I already have a method in my home automation system that knows when the house is occupied or not (a check every minute for certain mobile phones on the wifi network). I have this info currently stored in a text file and also a mysql database. It would be trivial to have the info stored elsewhere in different formats.

I like the working schedule to enable motion detection whilst I'm at work, but it would be great to have it work based on my occupied info. So I guess this is a feature request for an alternative method for telling motion to be detected?

ant-thomas, 2017-02-26:
Or another option, is there an API call of some sort to turn on motion detection? So I could integrate that into my "occupied" check.

ccrisan, 2017-02-27:
Nope, there's no such API as of right now. Plans are to implement one but that's not going to happen anytime soon.

closed 2017-02-27 by ccrisan

ant-thomas, 2017-02-27:
No problem. Thanks for the reply and thanks for the hard work!