motioneye-project/motioneyeos #766: Unable to Capture First Image

dmblack, 2017-02-13:
Hey Team,

Firstly; great work with this product. I have been using it some time now.

I'm having an issue where, after some time online; one of my two cameras will return 'unable to capture first image'. The only resolution so far; a reboot. Both cameras, initially; function as normal. It seems to degrade over time.

I do not have motion enabled; I'm just streaming the camera's (Currently using them as a secondary 'baby' monitor, actually).

I've had a bit more success after time with the cameras set to lower and lower resolutions (I'm now down to 320x240). This was demonstrated in the fact that sometimes, at higher resolutions; I would even have this error on startup.

Active user; so let me know what you need, and I'll get it!

bortek, 2017-06-01:
Closing this issue due to interest. If still interested please post your question in the forum page.

closed 2017-06-01 by bortek

dmblack, 2017-06-01:
Which forum? And what do you mean closing due to interest?

bortek, 2017-06-01:
Sorry a typo, I meant due to "lack of" interest. The forum is here

https://groups.google.com/forum/#!forum/motioneye