motioneye-project/motioneyeos #769: Notification email failed: check SMTP server name!

mickredirl, 2017-02-16:
Hi 
I am trying to setup email notifications but keep getting this error message each time:
 "Notification email failed: check SMTP server name!" 

The device I'm using is the Raspberry Pi3. I am running the Motioneye software version 20170212.

Before sending this email, I went through the list of known issues that are similar to this problem but found no solution. I even changed my password to take a symbol out of it which was this in brackets 
" * ". My password is not just letters and numbers... I know you updated software to accept symbols but did it to cross it off my list. I have uploaded the "motion.log" file, it appears as "motion log.txt". Hope it help. Just want to say, well done on creating the motioneye software.. Its brilliant.

Thanks :-)

I am using a Gmail account. Settings are as follows: 
SMTP server-->smtp@gmail.com
SMTP Port--->587
Use TLS---->Yes
Email Address--->myemail@gmail.com
From Address--->myemail@gmail.com
Send an email--->On
SMTP Account--->myemail@gmail.com
Attached Pictures Time Span--->2 seconds

[motion log.txt](https://github.com/ccrisan/motioneyeos/files/781480/motion.log.txt)

ccrisan, 2017-02-16:
`SMTP server-->smtp@gmail.com` - that's terribly wrong. Use `smtp.gmail.com`.

mickredirl, 2017-02-16:
Hi
Thanks for your help.. Really appreciate it :-)
I only noticed it after I had a break from the computer... So how deleted my first question??
Thanks again

closed 2017-02-17 by ccrisan