motioneye-project/motioneyeos #771: Sricam SP007 (SP012)

keplarQ3, 2017-02-17:
Hi - I have Motioneye OS version 20161212 running on a Raspberry 3.
I've successfully connected to a Wansview W1, Wansview W2 and a Linksys WV200. They all work just fine.

But I also have a Sricam SP007 onvif camera that I cant get to connect correctly despite trying just about every url combination in the world. I get the grey screen with "unable to open video device". However I do see 6.2/2.0 fps at the bottom of the window. - I attach a screenshot. I've even tried deleting the other cameras in case its a load issue to no avail.

Now strangely VLC in Ubuntu connects to the Sricam perfectly at rtsp://192.168.1.233:554/onvif1, and on my Android phone the Tinycam app also connects although it see's the camera as a Sricam SP012.

I'm at a loss and would appreciate any suggestions.

Logs also attached

Many Thanks - Doug

[image]

[motion.txt](https://github.com/ccrisan/motioneyeos/files/783146/motion.txt)

[motioneye.txt](https://github.com/ccrisan/motioneyeos/files/783158/motioneye.txt)

M3rl0z, 2017-05-12:
Maybe you've already solved ... this works with mine sricam sp007:  choose Camera-> RTSP/UDP Camera in the last row.
Bye, G.F.

[image]

bortek, 2017-06-01:
Closing this issue due to interest. If still interested please post your question in the forum page.

closed 2017-06-01 by bortek

marques576, 2020-04-19:
Can you give me your motion dettection settings?

starbasessd, 2020-04-19:
I don't have that camera.
I have noticed that my camera motion settings are in the 1% or less range
on most of my cameras. (640x480 and 800x600)
Auto Threshold Detection on for higher resolution
Auto Noise Detection On
Auto Despeckle Filter On
Light Switch Detection set to 0
If I mask it's usually set to Smart with slugishness set to 5
If I manually edit, I turn areas red that I don't want to trigger (I'm in a
flight path, so I mask most of the sky)
I use Show Frame Changes ON only to see what is triggering, then adjust
accordingly.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?