motioneye-project/motioneyeos #779: Samba USB Drive

48296, 2017-02-20:
Hello, 

I installed MotionEyeOS and added 2 webcams. I want to save the video files to an USB hard drive. For some reason the Samba shared network drive isn't poping up in windows 10. Do i need to edit the dilesin debug mode or is there any other fix. samba is turned on inside the gui.

ccrisan, 2017-02-20:
What is "dilesin debug mode"? Normally you should see the samba share in windows 10 w/o any issues or extra settings. Make sure you're accessing it using the correct path. It should look something like:

    \\your_motioneyeos_ip\

48296, 2017-02-20:
Woops, mistyped. 

I don't see the drive in the network and tried using the ip adres of the Pi. There is a issue with some of the same problems. https://github.com/ccrisan/motioneyeos/issues/706.

How can i solve the issue? I know the USB drive is mounted because I can record to it.

ccrisan, 2017-02-20:
What error message do you get when you access it using the \\ip\ syntax?

48296, 2017-02-20:
It's 0x80004005 . Before I had a retropie running with the same harddrive connected and shared by samba.
The network itself shouldn't be the problem.

48296, 2017-02-21:
I changed the workgroup. now it shows up in the network but is is giving an error when i try to enter it.

bortek, 2017-02-23:
And whats the error when you try to enter it ?

bortek, 2017-02-23:
This is duplicate of #706

closed 2017-02-23 by bortek