motioneye-project/motioneyeos #781: Continuous Recording with usb camera's

Loscarones, 2017-02-21:
Hi ccrisan,

Thanks for the good work! 
Although i have an issue. My setup is:
1x Raspberry pi 2
2x USB camera
network cable
network share saving

The issue is when i set one of the camera's to continuous recording the memory of the pi will slowly get full. I've set the video memory to 650mb. It looks like a memory leak, but i'm not the best at programming so i don't know. I hope you can fix this! 
[image]

bortek, 2017-02-23:
@Loscarones Can you elaborate a little bit. What do you mean by "memory of the pi will slowly get full". Are you using any specific tools?

shinji2009, 2017-02-26:
same here. pri3. everything was fine but then i swich to network share. sometimes it's looks loke memory leaked, sometimes it's don't. some network queue share problem maybe? help please

[image]

[image]

shinji2009, 2017-02-26:
oh noes now its something with local storage too. 
[image]
why git wont accept my logs or zip? i was able to get a 5 log files from expert settings. is there any sensative data in them?

bortek, 2017-06-01:
Honestly I dont know what your issue is. This is definitely  not a motioneyeos issue.  If still interested please post your question in the forum page. I will close this issue.

closed 2017-06-01 by bortek