motioneye-project/motioneyeos #782: using WebDAV disk and mount as network drive?

48296, 2017-02-21:
Because for some reason i can't get any smb network disks to work i figured i would like to use my stacks.
Stacks is a cloud storage that gives 1tb data for free. It can be mounted on any system, even on debian using some packages. 

Some of it is explaned here. https://www.transip.nl/vragen/756-stack-externe-back-voor-gebruiken/
It requires installation of packages, so I tried installing motioneye on raspbian. I got it working but for some reason it's missing some options in advanced mode. Also selecting "network drive" is not possible because it is missing. Is that normal? I would love to use motioneyeOS image but some essential bits are missing for my use and motioneye itself on top of raspbian isn't helping either.

Using STACKS would be ideal for anyone because it is stable, huge and free. Please consider adding it to motioneyeOS.

ccrisan, 2017-02-22:
Well webdav will definitely not make it to motionEyeOS so you'll have to make it work with Raspbian somehow.

Enabling network shares in motionEye is as simple as setting [smb_shares](https://github.com/ccrisan/motioneye/wiki/Configuration-File#smb_shares) to `true` in your motioneye.conf and make sure it runs as root.

Fast Network Camera is also not available on Raspbian, but I doubt you need it. What other functionalities do feel like are missing in pain motionEye but are present in motionEyeOS?

Philippe-M, 2017-02-23:
Smb is going to do very much to share on the local network, ftp for lan and wan but is not secure in the case of wan. Webdav is made to go through the wan.

ccrisan, 2017-02-23:
I admit that uploading to an external server via SMB is not recommended and often impossible. I have plans to add support for simple FTP, SFTP and HTTP upload services, but this is not a priority right now.

bortek, 2017-06-01:
Closing this issue due to interest. If still interested please post your question in the forum page.

closed 2017-06-01 by bortek

ccrisan, 2017-06-11:
https://github.com/ccrisan/motioneye/commit/296c29714ce2ab8deb0c9461b074ef39493b770a adds support for FTP uploading.