motioneye-project/motioneyeos #783: File retention logic

bobbydriver, 2017-02-22:
I set up motioneyeos to save files to a samba share on my network and retain them for 2 days. However after two days it deleted ALL the mp4 files on the Samba share older than 2 days. Sadly this included my entire video library which was on the same Samba share (but in a discrete directory). I have a backup so it’s no big deal to restore and I guess I need to set up a motioneyeos-specific Samba share going forward, but it would be good to make this clear in the screens how the logic works so that people don’t lose non-motioneyeos mp4 files in the process

molly78, 2017-02-23:
I know when I set up the keep files for X time, a pop-up dialogue box came up that warned about this...

ccrisan, 2017-02-23:
As @molly78 said, there's a big warning when enabling the file cleanup mechanism. The reason behind this is that motionEye has no way of knowing which videos were actually created by motion or were already there.

Ideally all media files generated by one camera should go to a directory that is specific to that camera and thus all files within it belong to that camera.

I'm sorry for your inconvenience but I wouldn't know how to make this better without too much effort.

closed 2017-02-23 by ccrisan

bobbydriver, 2017-02-23:
Thanks - I must have missed that warning the first time. I just changed the preservation time again and it did indeed warn me. Sorry - I must have been blind :)

johng2001, 2017-06-13:
Just had the same thing happen to me but with a USB hard drive that had about 50 GB of videos
in 4 or 5 folders.... overnight all the folders and contents not created by Motioneyeos where removed.
I didn't see a warning message when setting the keep for xx time (must be blind too)
can you please add a warning to the FAQ  about files being removed from 
storage devices automagically !!!