motioneye-project/motioneyeos #784: how righ Date-time stamp when router is offline - no internet connection possible

faltajan, 2017-02-23:
Hello ccrisan,

i have problem with right datetime stamp in system. As i understand it well, motioneyeos takes date and time from router. My problem is, my router is not online, and even router has wrong date time. I use router just for possibility to access camera system. Camera system is at place, where no internet connection is possible.
Is there any way how to setup right date time in motioneyeos?
System is shutting down every night and starting up evey morning...

Thank you for you help...

ccrisan, 2017-02-23:
motionEyeOS takes its date/time from google.com, not from your router. In the absence of an internet connection, you'll always end up with a date around 1970.

closed 2017-02-23 by ccrisan

rkbauer, 2019-10-12:
Hi, 
I really hope you had to provide an alternative option to setup the date method. I as well really like to set up the date of my motioneyeos (latest release 20190911 for RP4) via an (underwater) ethernet connection to my offline router that I'm using as access point (tp-link TL-WR802N). Unfortunately, I couldn't find a solution yet or any further information on how to configure the date method (neither via http or ntp). 
Any suggestions? I'd really appreciate your help on this !!!

alternatively: if i set up a date via a temporal wifi connection that is being removed, afterwards, will the date settings be lost afterwards if motioneyeos continues to run?

Thanks a lot!!!