motioneye-project/motioneyeos #785: loop recording enhancement request

shinji2009, 2017-02-23:
hi.
requesting for loop recording feature.
it is very useful, convenient and common feature for any surveillance hardware and software when using limited storage or a lot of cameras or standalone security device.

Movies - Preserve Movies setting now have only how many days should files be Preserved, this is very unconvenient for  surveillance  because nobody can know how many movies will be recorded per day - 1 or 100 and how soon  they will be needed. with  loop recording option files will be stored for maximum days possible.
thanks

Lama81, 2017-03-29:
Good idea here. I was looking for that feature too

bortek, 2017-06-01:
Closing this issue due to interest. If still interested please post your question in the forum page.

jasaw, 2019-01-15:
I think this feature is still worth implementing. Marking it as an enhancement.

DavHau, 2019-07-29:
Hi, i wrote a little script that adds the loop recording functionality. Take a look here: https://github.com/DavHau/motioneye-loop-record
Currently to use this script, you must upload it to your motionEyeOs system and enable its execution under 'Settings/File Storage/Run A Command'.
What do you think of merging this into motioneEyeOs and allowing the user to enable/disable this via checkbox?

jasaw, 2019-08-05:
@DavHau That's great. Just a few questions:
- would it be better if the script only looks for media files? Maybe look for mp4, mkv, avi, jpg?
- does it remove empty directories?

DavHau, 2019-08-05:
Currently it only skips files which start with '.' .
Since it is only looking in the output directory there should not be anything else than media files, correct? If not, I could add filtering, of course.
It does not remove empty directories. I will extend this functionality.
The thing i don't like so much about this solution is that each time the script decides to free space, all files in the directory are scanned to get the date. I will think about a solution to improve that. Inotify is not an option since it requires 1kb of RAM per file handle. 1000 recordings a day for one year is 350000 file handles, equals 350mb RAM.
Maybe reducing the frequency of deleting files by deleting more files at a time is the solution. If more space is cleared when deleting files, the less often deletion must be executed.

jasaw, 2019-08-06:
>Since it is only looking in the output directory there should not be anything else than media files, correct?

If it has the ability to delete all files, eventually someone will run this against a wrong directory. That someone might be me...

>Maybe reducing the frequency of deleting files by deleting more files at a time is the solution.

I think a good compromise here is to check for free disk space each time a video file is created. This won't pick up JPEG files because JPEG files are created a lot more frequently, so we also check free space every 15 or 30 minutes. Checking free space is a cheap operation because we only need to look at `df` output. To workaround frequently running the expensive operation of scanning every file for modified date, we set threshold with hysteresis, e.g. we only scan all files if disk usage is above 90%, and keep deleting until disk usage is down to 70%. We probably want to run this scan&delete process with the highest [nice](https://en.wikipedia.org/wiki/Nice_(Unix)) level so we don't choke the IO. We may even need to introduce some kind of delay in the scan&delete to avoid choking the IO.

>Assuming that media files are never modified by anything after creation

I think it's a fair assumption. Most people are interested in viewing the media files, and not modify them.

>Since motioneye is always creating one directory per day, the deletion algorithm could just delete whole directories at once and not single files.

We can't make this assumption unless motionEye starts restricting the output to directories.

DavHau, 2019-08-07:
Alright, i will add filtering for media files.
I'm not sure about the idea of not running this for JPEG files because,
1. Some setups might be configured to only record JPEG and nothing else. The function would never be executed
2. `df` is a really cheap operation an i think executing this often should not be a problem.

With the hysteresis idea i totally agree. This will reduce the amount of expensive operations dramatically. Also setting a low nice value makes a lot of sense.

I will think about a way on how to implement this into motioneye and make a pull request when i find some time.