motioneye-project/motioneyeos #786: Banana Pi D1 / Lamobo D1 Support

alanjames1987, 2017-02-24:
I see that MotionEyeOS supports the standard Banana Pi but I was wondering if the [Banana Pi D1](http://www.banana-pi.org/d1.html) or [Lamobo D1](https://www.kickstarter.com/projects/2081058999/lamobo-d1-develop-your-own-mini-camera) were supported and if anyone has experience working with them.

I've looked through the issues and all over the internet for an answer for this but I can't seem to find a definitive answer.

ccrisan, 2017-02-25:
The [Supported Devices](https://github.com/ccrisan/motioneyeos/wiki/Supported-Devices) page indicates all supported boards. So no, there's no support for the boards you mentioned.

See also https://github.com/ccrisan/motioneyeos/wiki/FAQ#can-motioneyeos-be-ported-to-insert-your-board-here.

alanjames1987, 2017-02-27:
I saw the supported devices page but I wasn't sure if it was extensive for just the devices that are regularly tested on.

How can I go about sending you a Banana Pi D1?

closed 2017-02-27 by alanjames1987

ccrisan, 2017-02-27:
I could give you my address for shipping. However after looking over the specs of the Banana PI D1 module, I'd recommend against porting and using motionEyeOS on this module. The specs are terribly low and the kernel/OS support is minimal, if not missing at all.

I believe motionEyeOS will run  terribly slow on this device, if porting it were be possible at all.

alanjames1987, 2017-02-27:
Understood. Thanks for the quick response.