motioneye-project/motioneyeos #787: Feature Request: multiple file storages

shinji2009, 2017-02-25:
hi. ok. having a muliple file storages will be so amazing. just for example because network storage can stop working due to network failure and in that case motioneyeos can use internal storage like /data/output/Camera1 until network become ok.

and i think network failure can cause motioeyeos stop working. this was made for two rpi3 and result is the same. rpi3s with V4L2 Camera.
i set up a network share storage. separate shared folders for each rpi. let motioneyeos transfer some videos to it, then turn off shared folder (on a win8.1 laptop using folder sharing properties). first, camera start to showing stiked logo when motion is detected, then picture goes back. after a few motion detections camera goes striked logo permanently. i turn shared folders back on. press on "test share" button and it says that "access network share sucseed!" but if press "open movies browser" button, there's no videos, like folder is emty. i reboot rpi. camera is still showing striked logo after reboot! it shows picture only after i adjust its fps settings. make it 5 then 10 again. ok. after a few seconds web interface stop working. i was  able to connect via ssh to see this
[image]
but can't do anything and just pull the plug. 
and i have a question is there a way to reboot a rpi3 with motioneyeos in that case using ssh? i was trying all commands i find in the web but none worked.

cuteboi, 2017-03-13:
Have you considered rsync on a cronjob that just deletes the source files on successful transfer or copy?

https://www.cyberciti.biz/faq/linux-unix-bsd-appleosx-rsync-delete-file-after-transfer/

shinji2009, 2017-03-13:
i was using a powershell script that connects to motioneye via ftp, download yesterday folder to laptop and delete source. some solution for chain recording.

closed 2017-06-01 by bortek