motioneye-project/motioneyeos #788: Unable to see Motioneye in windows explorer

Jeffon2wheels, 2017-02-25:
Thank you ccrisan for the great product.  I am really enjoying it.  Everything seems to work properly except that I can not see the raspberry pi in my windows explorer under network?  Any ideas what could be going wrong?

Jeffon2wheels, 2017-02-25:
Log attached I think

[Motioneyelog.txt](https://github.com/ccrisan/motioneyeos/files/801343/Motioneyelog.txt)

Jeffon2wheels, 2017-02-25:
Motion log also attached if that is helpful
[motion.txt](https://github.com/ccrisan/motioneyeos/files/801344/motion.txt)

ccrisan, 2017-02-25:
What error message do you get when you try to access it from Windows Explorer? Have you correctly identified the IP address of your motionEyeOS? Does the path you're accessing look something like:

    \\your_motion_eye_ip_address\

?

Jeffon2wheels, 2017-02-25:
Thanks for getting back to me so quick.  I do have the correct IP because I can open a browser window and view the camera.  I do not get an error message.  I thought that when you open "networks" in windows explorer that the Pi would be there along with the other computers and devices on my network?  I may have missed a step if that does not happen automatically.

I am trying to have the video and images stored on my windows pc instead of the sd card.

ccrisan, 2017-02-25:
With Windows Explorer and network share there's always this story about workgroups. If the two computers don't have the same workgroup, one might not see the other listed as expected. To be honest, this is too complex for me.

Anyways, please try to access it as indicated above and see if it works.

bortek, 2017-02-25:
I added this topic to FAQ section (https://github.com/ccrisan/motioneyeos/wiki/FAQ#unable-to-see-motioneye-in-windows-explorer-under-network). Not the first time people ask about it. This is actually not a motioneye specific question but rather a windows network related one. So a simple googling on the topic would yield  more results.

closed 2017-02-25 by bortek