motioneye-project/motioneyeos #799: video darker than still images

shinji2009, 2017-02-27:
hi
video for some reason much darker tnan pictures
here is a still image
[image]
here is a video snapshot taken from same time video without any brightness/contrast adjustmets 
[image]
rpi3 20170212

ccrisan, 2017-02-27:
Which pictures?

shinji2009, 2017-02-28:
pictures that a taken by still Images setting.  and that are on a desktop.  as i understand it is fast changing pictures on a desktop as a camera preview, not a video feed

ccrisan, 2017-02-28:
If this is a Raspberry PI CSI camera, it may have to do with video vs stills mode. And then it's not a motionEye-related issue.

shinji2009, 2017-02-28:
well if raspberry PI CSI camera can output two streams at the same time - still image and video. i was thinking that camera produce only series of still images, and motioneye make a jpegs and/or video of them using gpu.
it is still pity that desktop feed does not show as video is recorded in the darkness. on the desktop there is something but in actual video file there is nothing.
sorry

shinji2009, 2017-03-03:
try this with old 320*240 usb camera, i think it can't do two streams at a time - same thing. looks like still images and desktop are +0.2 gamma.

shinji2009, 2017-03-03:
@ccrisan can you please consider once more #797 ?
is there a problem for motionos to convert pixels into percents by itself? it's os О_о

ccrisan, 2017-03-03:
@shinji2009 motionEye does not handle the overlay, nor those settings. motionEye exposes the threshold setting as a percent (it's easier for most users than having to deal with a big number) but internally it transforms it to the threshold number required by motion.

closed 2017-03-03 by ccrisan

shinji2009, 2017-03-04:
this issue is not about percents, are you close it by mistake?

ccrisan, 2017-03-04:
I closed it because I have no idea what to do about it.

shinji2009, 2017-03-05:
@ccrisan is there any chance that  somewhere used a key -optimize_dark_images for the jpeg library or -gamma-minus-0.2 for video encoder?