motioneye-project/motioneyeos #800: Sendmail Problem

peternsomerset, 2017-02-28:
Hi, 
I am having problems sending a test email and have been trying to do it on both my gmail.com account (I have changed my account to use less secure apps) and my virginmedia.com account which give the same results:
2017-02-28 14:57:16: [motioneye]    ERROR: notification email test failed: Connection unexpectedly closed: timed out
Traceback (most recent call last):
  File "/usr/lib/python2.7/site-packages/motioneye/handlers.py", line 835, in test
  File "/usr/lib/python2.7/site-packages/motioneye/sendmail.py", line 53, in send_mail
  File "/usr/lib/python2.7/smtplib.py", line 256, in __init__
  File "/usr/lib/python2.7/smtplib.py", line 317, in connect
  File "/usr/lib/python2.7/smtplib.py", line 365, in getreply

I have checked passwords/servers but still cannot get a message. Can someone please help and point me to information to get this sorted as its driving me nuts!? 
Thanks

### Issue Already Reported
First of all, search through the [list of issues](https://github.com/ccrisan/motioneyeos/issues); someone else might have already reported your issue. If you found it, feel free to contribute to the issue with comments.

### Report New Issue
If your issue hasn't been reported yet, [report it yourself](https://github.com/ccrisan/motioneyeos/issues/new).

If you're using motionEye on a Linux distro rather than motionEyeOS, please switch over to

peternsomerset, 2017-02-28:
Sorry  - just found I had port 465 for gmail changed to 587 and its working.

closed 2017-02-28 by peternsomerset

freddy64, 2018-03-20:
Hello, 
Can I now disable the " Less Secure Apps” in Gmail and still get notifications?
I needed to permit logins from "Less Secure Apps" to get notification from motionEyeOs but google tells me this is a security problem. So I'm wondering how much of an issue this is.
 I'm using motionEyeOS 20180224 with Raspberry Pi Zero W as security cameras.
Thanks