motioneye-project/motioneyeos #801: Cannot Download motioneyeos-raspberrypi3-20170212.img.gz page not found...

Osirus126, 2017-02-28:
PLEASE DO CREATE ISSUES WHICH ARE NOT RELEVANT TO THIS PROJECT. 
QUESTIONS LIKE "what is the best codec for video" or "Which is best Pi2 or Pi3" ARE NOT RELATED 
TO THIS PROJECT AND SHOULD BE ASKED ELSWHERE e.g. ON RASPBERRY PI FORUMS.

### Issue Already Reported
First of all, search through the [list of issues](https://github.com/ccrisan/motioneyeos/issues); someone else might have already reported your issue. If you found it, feel free to contribute to the issue with comments.

### Report New Issue
If your issue hasn't been reported yet, [report it yourself](https://github.com/ccrisan/motioneyeos/issues/new).

If you're using motionEye on a Linux distro rather than motionEyeOS, please switch over to [motionEye's issue tracker](https://github.com/ccrisan/motioneye/issues/).

##### Version And Hardware

Please clearly specify what OS version you are running and on what board. It is also important to know what types of cameras you have attached to your motionEyeOS system.

##### Attach Log Files
In the *Expert Settings* section of the settings panel you will find a few log files that can be downloaded and saved locally to your computer. These log files can help identifying the cause of your issue. You can attach these files to the body of the issue (or to a comment).

##### Attach Screenshot
Don't be afraid to attach some screenshots, if you think your problem can be more easily explained visually. Images can be attached directly to the issue on github.

##### Sensitive Data
Never attach configuration files (the ones found in `/data/etc`) before removing sensitive data from them (e.g. passwords, public IP addresses, etc).

-------------------------------------------------------------------------------------------------------------------------------------------

Trying to download motioneyeos-raspberrypi3-20170212.img.gz but i keep getting timed out or page not found.. if you could help me that would be great!

mcjohnffs, 2017-02-28:
Same problem here. Even the earlier releases wont download

fersingb, 2017-02-28:
Looks like there are currently some issues with amazon S3

ccrisan, 2017-02-28:
Yeap, they seem to have some serious problems. I'll close this issue now as it's not ours.

closed 2017-02-28 by ccrisan