motioneye-project/motioneyeos #802: Feature: Is there any chance to make the Web Interface on IPV& available?

chpflueger, 2017-03-01:
Motioneyeos is a wonderfull Produkt. 
Unfortunaly my Internet Service Provider provides only DS Lite. 
That means that i hava only a chance to connect my external Serives via IPv6. 
Is there any chance to bind your WebInterface to ALL Interfaces (especially the IPV6 addresses)

Kind Regards
Christian

PLEASE DO NOT CREATE ISSUES WHICH ARE NOT RELEVANT TO THIS PROJECT. 
QUESTIONS LIKE "what is the best codec for video" or "Which is best Pi2 or Pi3" ARE NOT RELATED 
TO THIS PROJECT AND SHOULD BE ASKED ELSWHERE e.g. ON RASPBERRY PI FORUMS.

### Issue Already Reported
First of all, search through the [list of issues](https://github.com/ccrisan/motioneyeos/issues); someone else might have already reported your issue. If you found it, feel free to contribute to the issue with comments.

### Report New Issue
If your issue hasn't been reported yet, [report it yourself](https://github.com/ccrisan/motioneyeos/issues/new).

If you're using motionEye on a Linux distro rather than motionEyeOS, please switch over to [motionEye's issue tracker](https://github.com/ccrisan/motioneye/issues/).

##### Version And Hardware

Please clearly specify what OS version you are running and on what board. It is also important to know what types of cameras you have attached to your motionEyeOS system.

##### Attach Log Files
In the *Expert Settings* section of the settings panel you will find a few log files that can be downloaded and saved locally to your computer. These log files can help identifying the cause of your issue. You can attach these files to the body of the issue (or to a comment).

##### Attach Screenshot
Don't be afraid to attach some screenshots, if you think your problem can be more easily explained visually. Images can be attached directly to the issue on github.

##### Sensitive Data
Never attach configuration files (the ones found in `/data/etc`) before removing sensitive data from them (e.g. passwords, public IP addresses, etc).

ccrisan, 2017-03-01:
There are no plans to add support for IPVv to motionEyeOS anytime soon. Sorry.

MeisterGig, 2018-10-04:
Most of the home networks are using DS Lite Tunnels and only have IPv6. I think it should be on the agenda to add IPv6 support.

ccrisan, 2018-10-04:
Alright, we could consider enabling IPv6 on motionEyeOS.

afedulov, 2019-11-14:
Please, enable IPv6 support - there is simply no way to access Raspberry Pi from the outside on major German ISPs without switching the provided modem/router into bridge mode and buying a separate router. This is a major blocker. And more and more providers everywhere are switching to DS-Lite, which makes access via IPv4 impossible.
P.S.: Do I get it correctly, that IPv6 is supported when installing motioneye on Raspbian instead of using mitioneyeos? 
P.P.S.:  Is this [patch](https://github.com/ccrisan/motioneyeos/blob/c296914e4aa7a5a949e7286ea769f9125e824859/package/lighttpd/0001-Fix-default-config-file.patch) the one responsible for disabling IPv6 or is there something else that might need to be reverted? 
Thanks.

afedulov, 2019-12-02:
@ccrisan  Thoughts/ideas/comments/tips?

ccrisan, 2019-12-02:
@afedulov I can try and enable IPv6 in the kernel and OS in general, but I won't update all the scripts and UI helpers that deal with IP configuration so that they work with IPv6.

afedulov, 2019-12-02:
@ccrisan this would be nice, thank you! I personally use a USB camera, so I am not even sure I will need to deal with any of those.

ccrisan, 2019-12-15:
https://github.com/ccrisan/motioneyeos/commit/9d2febd692adf9a06cf95b428d5563d5261f3dd7 removes the `ipv6.disable` argument from kernel commandline. It's possible though that other parts of the OS require changes as well.

Goldglv, 2020-09-03:
I'm running into an issue with my Xfinity Xfi router that will NOT allow port forwarding unless the device is using ipv4,  What do I need to add in the settings.py file to disable ipv6?