motioneye-project/motioneyeos #804: Built-in WiFi not working on the new Raspberry Pi Zero W

jawsper, 2017-03-01:
It seems the built-in WiFi doesn't work yet on the RPi Zero W.

I tested it with the most recent Raspbian and it works fine on there.

jawsper, 2017-03-01:
This is on the most recent version of MotionEyeOS (20170212).

ccrisan, 2017-03-01:
How the heck did you get your hands on a RPi Zero W already? You luck bastard.

Anyways, as far as I read in the news, the chip is the same as that used by the RPi3, which is brcmfmac. Can you connect a monitor and see what the boot log has to say about it?

jawsper, 2017-03-01:
On a hunch, I copied the bootcode.bin/fixup.dat/start.elf from the raspbian jessie image, and that seems to do the trick!

So I guess a simple firmware update will get the WiFi working on the Pi Zero W.

(I ordered it the moment I saw the post on Hackaday about it, and I apparently beat the crowd :))

ccrisan, 2017-03-01:
Ah, most likely the firmware needs to be updated for the W :)

jezmck, 2017-03-02:
A fresh install doesn't work though - it reboots after failing to find a wired network?

jfornango, 2017-03-12:
I've been playing with the Zero for a few weeks now. As soon as I heard the W was out, ran to my local MicroCenter and snagged one.

Running the latest download of Jessie (2017-01-11), I am having the same issue. Jessie can't find the WiFi or Bluetooth.

As a check, I pulled the SD and put it in my Zero, which has a RedBear IoT pHAT 1.1 mounted on the GPIO. It picked up the hardware immediately, I'm typing this post with near-zero config work. (disabled overscan in /boot/config.txt, and set WiFi credentials)

I would definitely appreciate any thoughts on where to look.

jezmck, 2017-03-12:
There is a working solution a few posts up in this thread. No *official*
support yet though.

jawsper, 2017-03-12:
@jfornango That is most certainly not a recent enough version of Debian Jessie. The one I downloaded after getting the Pi Zero W is the 2017-02-16 version, and currently the latest version is 2017-03-02. I had immediate success with the February version, so best to download the more recent update and try again with that.

On another note, the firmware in this repository has now been updated, so when @ccrisan releases a new build, or if you build it yourself, motioneyeos should work just fine on your Pi Zero W.

jfornango, 2017-03-12:
Wow...  three releases in two months.  Ok, I guess I'll need to grab the new one.
In the meantime, I did find a workaround. (for anyone who is into wrestling down issues the hard way... like me.  XD )

So, I have a Pi Zero with a RedBear WiFi/BT card.  I moved my Pi Zero-W SD card to the Zero. (other folks have done this with a Pi 3)  It picked up the Wifi with no problems.  From there, did the following:
1. sudo su -
2. apt-get update
3. apt-get upgrade
4. rpi-update

Shut down the Zero, and moved the SD card to the Zero-W.
It picked up the WiFi instantly, though the BT was still out.
From there, another apt-get update and reboot got the BT up and running without any issue.

benmeryon, 2017-03-13:
Hi !

I received my Pi Zero W on saturday. I tried the latest Pi Zero version of motioneyeos but the wifi seems to not work. I copy the release with imagewrite.sh script. All is good but I can't connect to the Pi.
Is there a tip I miss ? Do I have to wait for an update version of motioneyeos?
I'm rookie on that !

Thanks a lot. :)

wdawson001, 2017-03-13:
@benmeryon The repository has been updated for the Pi Zero W, but not been officially compiled. 

I've put together an unofficial release based on a recompiled repository:

benmeryon, 2017-03-13:
Thank you !
But, do I have a Raspberry B to configure the wifi ?
_* use a model B to boot it, configure the wireless network in the settings panel and boot the SD card on your model A/A+/Zero_ (on **4.**)

petermeigs, 2017-03-14:
Everyone, thanks for your comments as they helped me a lot.  I had a lot less trouble getting things to work that was described so I thought I'd let you know what worked for me.

1. I had a copy of rasbian an SD that I had prepared and had been using on a non-w  rpi zero.
2. I put this in my new rpi-zero-w and sure enough: no wfi and no bluetooth.
3. I have a wifi dongle that I was using on my non-w rpi zero and placed that in a usb hub along with my  wireless keyboard/mouse dongle.  When booted, that connected just fine. 
4. I then did the steps mentioned above: sudo su-; apt-get update; apt-get upgrade; rpi-update.
5. I then shut it down with poweroff and removed my wifi dongle.  Then I powered it back on.
6. Now both the bt and the wifi of the rpi zero worked.

If you recently ran  rpi-update on a different rpi than the one booted booted from, it may think you already did the rpi-update and decline to do it.  If you have this situation, you can delete the file where the checksum is stored (I forgot where that is, sorry) or take a look here: https://tech.enekochan.com/en/2014/03/08/upgradedowngrade-to-a-specific-firmware-kernel-version-with-rpi-update-in-raspbian/ to get to a specific version.  I didn't have to do this because there was newer firmware than I had installed using that sd card before.  I have run into that situation before when using the same sd on multiple rpi's that I was trying to prepare for use.

I believe rpi-update updates the rpi zero's on-board firmware so would not really have any effect on the sd card except to capture the checksum.  Since it looks like rpi-update makes a remote access to get the firmware, I believe you need to have an internet connection for the rpi-update to work.  My wifi dongle gave me that so I'm guessing that's why it worked so easily for me.

I'm pretty new at this rpi stuff so if I've guessed incorrectly, I'm happy to be corrected.

I hope this helps someone...

wilinsky, 2017-03-16:
When I connected my Pi Zero W to a monitor when it got to the wireless spa_supplicant section of states it can't find a wireless device, goes to connect wired and states no wired network and then reboots.

spieiga, 2017-03-16:
@wilinsky Are you running motioneyeos 20170212 release?  If so, I believe you need to update the rpi firmware.

bananosky, 2017-03-17:
@petermeigs Thanks for your suggestion. It worked for me with a slight difference. I have my other Pi Zero with a network dongle. One more note is that I didn't have to run rpi-update. Basically, this is what I did
1. Install raspbian on Pi Zero W (I'm using piaware)
2. Move the network dongle from Pi Zero to Pi Zero W. Have HDMI monitor connected to Pi Zero W and boot it. 
3. I connected the screen to see which IP the DHCP was going to assign. If you don't have a monitor you could find out the IP from your DHCP server.
4.  Update Pi Zero W:  sudo apt-get update; sudo apt-get upgrade; reboot
5. Login and wlan0 showed up. Configure wlan0, and shutdown. 
6. Remove network dongle and startup the Pi.

petermeigs, 2017-03-17:
Hmm, I had convinced myself that it was running the rpi-update network connected was what did the trick.  Now it seems that running apt-get update and apt-get upgrade network connected is what does it instead. 

This makes some sense because I have been testing the Adafruit_DHT Addsong AM2302 temperature humidity sensor and found that I need to rerun setup.py when I move an SD card from an rpi3 to the rpiz-w.  It seems it is sensitive to the rpi version and 3 is different from zero. I'm now suspecting that the rpi-upgrade does something similar.  

I have been planning on getting several rpiz-w's (Is that an ok nickname? If not, any suggestions?) and distributing them about the home to be able to capture and record temp and humidity in rooms.  My home is not that big but it heats unevenly.  Anyway, I had hoped to be able to use chef to build and maintain consistent images on the rpiz's but having to plug and unplug hardware will interfere with that.   (See http://blog.ittoby.com/2016/04/installing-chef-on-raspberry-pi-23.html)

I testing running the rpiz-w's sd card in a rpi3 and that connected to the wlan with no issues.  

To start off my chef process, I had to create a tweaked img file that started up with ssh turned on and a default wlan0 connection.   I suppose I could retweak my base img by running apt-get update, apt-get upgrade and then capture a single img that works for both.  

The reason behind all of this for me (managing my rpi's with chef) is getting a bit off-topic for this thread but the mystery I see now is why a wifi that built on a sd in rpi3 won't work on a rpiz-w. 
 
BTW, the hubpiwi blue (http://referencedesigner.com/blog/hubpiwi-blue-usb-hub-wifi-and-bt-for-pi-zero/4343/) works out of the box. This item only makes sense for a non-w rpi-zero.   If you don't need the USB hub though, the rpi-zero-w is a much tidier and a few bucks cheaper solution

BTW-2, I'm not running motioneyeos at all.  The problem for me was rpi-zero-w not having a wlan0 out of the box.  It looks interesting.  Thanks for the tip here.

wilinsky, 2017-03-18:
Just to be clear. I can get the wifi working on the Pi Zero W with Jessie Lite. Motioneyeos does not see the wireless device on the Zero W.

wdawson001, 2017-03-18:
@wilinsky try this solution for the pi zero: https://github.com/ccrisan/motioneyeos/issues/804#issuecomment-286071303

It's recompiled using the new Jessie.

gustamucho, 2017-03-18:
Thanks wdawson001, I followed your comments and tried this solution on my newly acquired Raspberry Pi Zero W with the official raspberry Pi Noir camera with success.  Special thanks to ccrisan on all the hard work and effort put into this project.  As a noob,  this project has enhanced my wireless home security.

wilinsky, 2017-03-18:
@wdawson001  Success! Your build is working great.

italianjoe69, 2017-03-19:
@wdawson001 - Thank you for your build.  It works great.  Any ideas on how to get SSH working? 

@ccrisan - Thank you for all the work you've done with motioneyeos.  Donation on its way!!!

ccrisan, 2017-03-20:
@italianjoe69 thanks :)

The new official version (with support for the Pi Zero W) is on its way.

ciavolella, 2017-03-21:
@wdawson001 - Everyone seems to be able to get this to work except me.  I'm using an 8Gb micro SD and Win32DiskImager.  The downloaded img file integrity appears to be intact, as I can open and view with 7zip.  I've put the wpa_supplicant.conf file in the boot partition, but when I put the thing in my Pi Zero W, the ACT LED blinks green once, then stays off.  I'm using a 1000mA power supply, and I've tried the 2A one is use with my Pi3, but nothing changes.  It won't boot at all on the Pi3, but that's likely due to it not having the correct boot files in the boot partition.  Help, anyone, please.

@ccrisan - A second thanks for making an easy to use OS version.  Now I just have to get it to work.  Any date target for the Pi Zero W version?

italianjoe69, 2017-03-21:
@ciavolella - Any chance you can hook up a monitor and see what is happening?  That is what I had to do in order to troubleshoot my initial attempt.  I bought the HDMI adapter at Best Buy.

ciavolella, 2017-03-21:
@italianjoe69 I get a bunch of errors about post version upgrading failed (2015 and 2016 builds) no motioneye.conf file failure, some other junk, and then it dumps me to a login screen that I can't get past.

italianjoe69, 2017-03-21:
@ciavolella - Are you using the image and instructions @wdawson001 posted [here](https://github.com/ccrisan/motioneyeos/issues/804#issuecomment-286071303)?  Make sure to use the image referenced [here](https://1drv.ms/u/s!AmNBhv72BI68got82f3cirzYG0bfRw).
I followed this exactly and had no issues.

wdawson001, 2017-03-21:
@ciavolella I would suggest starting over - just to make sure nothing has gone wrong

Download a fresh image file from [here ](https://1drv.ms/u/s!AmNBhv72BI68got82f3cirzYG0bfRw) (No extraction needed). Using win32diskimager to write the file to the SD card. Then add your custom wpa_supplicant.conf to the boot partition. 

Once done, boot up connected to a screen to see what happens. Then let us know if you have an issues.

ciavolella, 2017-03-21:
@wdawson001 @italianjoe69 - That did it.  Not sure why my initial download/write failed.  Everything _appeared_ to be working... Anyway, I should have figured - "just do the exact same thing you did last time, and expect different results" rule when working with Linux.  It's up and running now, thanks!

nickolas1969, 2017-03-24:
RPi0W, Jesse lite 2017-03-02 image, wireless worked, used raspi-config to change my locale from GB to US, rebooted.. no wifi. :-/ (wlan0 shows up, but doesn't associate to my home network anymore) 

Dmesg has brcmf red errors "not a ISO3166 code"???

ccrisan, 2017-03-26:
https://github.com/ccrisan/motioneyeos/releases/tag/20170326 adds full support for the Zero W.

closed 2017-03-26 by ccrisan

agilob, 2017-04-08:
@ccrisan will it update support for Rpi Zero W in raspbian repo? How long before it's updated there?

ccrisan, 2017-04-08:
@agilob motionEyeOS has nothing to do with Raspbian. Therefore this update won't affect Raspbian at all. If you're looking for motionEye on Raspbian, see https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian

nickolas1969, 2017-04-08:
Whoops,

I am sorry, I did hit a thread that had something I was fighting but you are correct,  it's the wrong forum for it.

Nick

Nick Ellson - from iPhone (forgive typos)
CCIE #20018; VCIX-NV, CNSE
Network Hobbyist
"Educating Layer 8, one user at a time."

agilob, 2017-04-08:
@ccrisan I just upgraded packages on raspbian Zero-W with `pip install motioneye --upgrade` and motioneye is unable to start the server (same as before upgrade). It's enabled by systemd and started (main process in `top` table), it's not logging anything to `/var/log/motioneye` which has chmods=777, looks like motioneye is just restarting itself in a loop:
```
-- Logs begin at Sat 2017-04-08 19:35:44 UTC, end at Sat 2017-04-08 19:41:45 UTC. --
Apr 08 19:35:50 motioneye systemd[1]: Starting motionEye Server...
Apr 08 19:35:50 motioneye systemd[1]: Started motionEye Server.
Apr 08 19:35:56 motioneye systemd[1]: motioneye.service: main process exited, code=killed, status=4/ILL
Apr 08 19:35:56 motioneye systemd[1]: Unit motioneye.service entered failed state.
Apr 08 19:35:57 motioneye systemd[1]: motioneye.service holdoff time over, scheduling restart.
Apr 08 19:35:57 motioneye systemd[1]: Stopping motionEye Server...
Apr 08 19:35:57 motioneye systemd[1]: Starting motionEye Server...
Apr 08 19:35:57 motioneye systemd[1]: Started motionEye Server.
Apr 08 19:36:01 motioneye systemd[1]: motioneye.service: main process exited, code=killed, status=4/ILL
Apr 08 19:36:01 motioneye systemd[1]: Unit motioneye.service entered failed state.
Apr 08 19:36:01 motioneye systemd[1]: motioneye.service holdoff time over, scheduling restart.
Apr 08 19:36:01 motioneye systemd[1]: Stopping motionEye Server...
Apr 08 19:36:01 motioneye systemd[1]: Starting motionEye Server...
Apr 08 19:36:01 motioneye systemd[1]: Started motionEye Server.
Apr 08 19:36:03 motioneye systemd[1]: motioneye.service: main process exited, code=killed, status=4/ILL
Apr 08 19:36:03 motioneye systemd[1]: Unit motioneye.service entered failed state.
Apr 08 19:36:04 motioneye systemd[1]: motioneye.service holdoff time over, scheduling restart.
Apr 08 19:36:04 motioneye systemd[1]: Stopping motionEye Server...
Apr 08 19:36:04 motioneye systemd[1]: Starting motionEye Server...
Apr 08 19:36:04 motioneye systemd[1]: Started motionEye Server.
Apr 08 19:36:06 motioneye systemd[1]: motioneye.service: main process exited, code=killed, status=4/ILL
Apr 08 19:36:06 motioneye systemd[1]: Unit motioneye.service entered failed state.
Apr 08 19:36:06 motioneye systemd[1]: motioneye.service holdoff time over, scheduling restart.
Apr 08 19:36:06 motioneye systemd[1]: Stopping motionEye Server...
Apr 08 19:36:06 motioneye systemd[1]: Starting motionEye Server...
Apr 08 19:36:06 motioneye systemd[1]: Started motionEye Server.
Apr 08 19:36:18 motioneye systemd[1]: motioneye.service: main process exited, code=killed, status=4/ILL
Apr 08 19:36:18 motioneye systemd[1]: Unit motioneye.service entered failed state.
Apr 08 19:36:18 motioneye systemd[1]: motioneye.service holdoff time over, scheduling restart.
Apr 08 19:36:18 motioneye systemd[1]: Stopping motionEye Server...
Apr 08 19:36:18 motioneye systemd[1]: Starting motionEye Server...
```

I followed custom config from link you posted,`raspistill -o img.jpg` produces correct image. Any advice where to look?

ccrisan, 2017-04-09:
@nickolas1969 this repo still isn't about Raspbian + motionEye. Nevertheless, I'll try to answer your question.

Raspbian uses `systemd` and `systemd` comes with `journald`, which takes care of all system logging. If you're looking for some logs that come out of motionEye, the following command will help:

    journalctl -u motioneye

If you want it stored into a file, just run:

    journalctl -u motioneye > motioneye.log

braindumped, 2017-04-14:
downloaded 2017-04-10-raspbian-jessie-lite, flashed and supplied a wpa_supplicant.conf to the boot partition.
my zero-w does not connect to wifi with its own wifi 
BUT when I put in a usb-wifi-dongle it connects to my AP without problems
What could be wrong?

ccrisan, 2017-04-16:
@braindumped you have downloaded a Raspbian OS image while you're on the issue tracker of motionEyeOS, which is a totally different OS. Please look for help on Raspbian/Raspberry PI forums.

braindumped, 2017-04-16:
argh. sry. 
thanks!

rrmills7777, 2017-05-09:
I know this issue is closed, but I'm having some issues.  I have a Pi Zero (non-wireless version) with a HubPiWi.  MotionEyeOS is not working with this out-of-the-box, but raspbian does.  Anybody have any tips?

Steve2Q, 2017-12-14:
Just FYI on how I handled the PiZeroW wifi issues. First I tried multiple times using Windows to do the install as instructed. The PiZero always looked for a wired LAN and would just do a reboot loop. I resolved the problem 2 ways:
1. Installed Raspian Stretch on the card and followed the instructions on installing Motioneye with Raspian. This worked fine, except the Pi then acts as a hub and some of the configuration items will not be available. This works because you configure the wifi thru Raspian.
2. I took a USB stick and put on the Live persistent version of Ubuntu. I then used this to boot my computer into Ubuntu. Using this I then followed the "normal" instructions on placing the Motioneye image on the card (most recent version), and then added wpa_supplicant.conf to the card using GedEdit.
Put the card in the PiZeroW, and it booted right up with the wifi working and all configuration items present.
I don't know why this worked...maybe the new version of Notepad++ has a bug??

Hope this helps someone.

Steve

DasFatBear, 2018-04-24:
How I fixed this issue (and this thread was a great help) as I had it happen after I had previously configured it in MontionEyeOS:

1.  Boot the device with an Ethernet dongle so it can pull an IP from your DHCP server, or from the preconfigured static IP (if you're configured that way).

2.  I connected a wireless keyboard and monitor to the Zero W and logged in at the command prompt using root and the admin password from the GUI.

3.  Change to the /etc/data/ directory 

4.  Remove the offending old wpa_supplicant.conf file (in my case):
      rm -f wpa_supplicant.conf

5.  Reboot. 

During the reboot, it should read from the wpa_supplicant.conf file in the /boot section.