motioneye-project/motioneyeos #807: Raspberry Pi Zero W cannot find Local Camera

jezmck, 2017-03-02:
Hardware confirmed working with vanilla latest Raspbian OS.

Am using a USB WiFi dongle for the time being.

Add camera > local camera > (no cameras)

I'm not sure what else I can do - any suggestions?

30dbs, 2017-03-03:
Can confirm the same issue, if I switch to Raspbian SD card it works fine, so the camera is definitely plugged in correctly.

I followed Jawsper's suggestion in https://github.com/ccrisan/motioneyeos/issues/804 to get the WiFi working properly but cannot select a camera one I log into the web interface.

kirbyan, 2017-03-03:
I can confirm the same behaviour - Official Raspberry Pi Camera not working on Raspberry Pi Zero-W with MotionEyeOs but works fine with latest Raspbian SD Card.

jasharrar, 2017-03-03:
Hi also having similar issues:
New Pi Zero W
NOiR camera v1.3

cant get the wifi working so using a dongle which connects first time
but main issue is says no camera connected. everything worked with jessie, is there a way to copy the relevant files across?

many thanks for any help

jasharrar, 2017-03-03:
also @30dbs how did you open bootcode.bin (win 10)

30dbs, 2017-03-03:
@jasharrar I copied it on my Linux desktop, but as far as I am aware it should work fine on windows 10 as the files were in the boot partition which is fat32.

If you couldn't see the file maybe maybe you have systems files hidden?

After copying and rebooting the wifi worked fine.

jasharrar, 2017-03-03:
@30dbs i found the file, i just cant open it lol. thanks ill have a go on a friends laptop.

30dbs, 2017-03-03:
@jasharrar. you don't need to open it, just copy the three files 

bootcode.bin
fixup.dat
start.elf

From the boot partition of jessie to the boot partition of motioneyeos

jasharrar, 2017-03-03:
thanks @30dbs i have tried that but im still getting stuck in the boot loop due to no device on wpa_supplicant.  i have copied the files from jessie lite and my own wpa conf file.

jezmck, 2017-03-03:
Copying those files worked for me, thanks @30dbs, though there's still no camera :(

coatezy, 2017-03-04:
No camera here either. rev 1.3 camera.

irishbiker, 2017-03-05:
I copied the three files from a full fat version of Jessie but upon booting motioneyeos it is looking for a wired network - fails then reboots.  Can anyone give me some pointers

30dbs, 2017-03-05:
@irishbiker

You also need to create a text file in the boot partition called...

wpa_supplicant.conf

With the contents below...

update_config=1
ctrl_interface=/var/run/wpa_supplicant
network={
scan_ssid=1
ssid="ssidname"
psk="password"
}

irishbiker, 2017-03-05:
@30dbs 

Thanks - I added the file using both notepad++ and notepad but still reports only checking for the wired network :/

jasharrar, 2017-03-05:
@irishbiker I got mine working finally, I did a fresh install, with the conf file as above and the 3 files from Jessie(but I found that using the full version worked not the lite one for some reason). 

Has anyone got any closer getting the on board camera working?

jsmacdou, 2017-03-07:
I have done several fresh installs, with proper wpa supplicant config and the three files from Jessie (both lite and regular) but I am met with a "no device" message for wifi on boot. Any ideas?

ccrisan, 2017-03-07:
Guys, there's no support for Raspberry PI Zero W in motionEyeOS yet. The next version will add support for Zero W.

Twanfox, 2017-03-07:
Heh. While there's no support right now, I did manage to get it working. Changes made in addition to above:

- edit config.txt in /boot to include the line start_x=1
- copy fixup_x.dat/start_x.elf to /boot from Pixel-based Raspbian image, similar to fixup.dat/start.elf

I wound up copying a few other files over during testing, but I'm not sure their purpose, so I can't say they're related. Once I caught on that the camera is now activated by start_x in the config.txt file, that seemed to be the focus. For me, this resolves the on board camera issue. 

Sorry, ccrisan, if it's jumping the gun. I know I like a good challenge, even if I'm not an expert by any means.

wdawson001, 2017-03-07:
Hi all, (especially ccrisan - awesome work).

I've built the latest img file from source (as per your [instructions](https://github.com/ccrisan/motioneyeos/wiki/Building-From-Source)) and it is working very well for my pi zero w - tested on 2 Pi Zero W's with v1 camera and version 2.1 noIR camera. 

@ccrisan it ok to share the img file?

ccrisan, 2017-03-07:
Sure thing, just share the link with us. I'll receive my board soon and I'll be able to run the tests myself.

wdawson001, 2017-03-07:
Cool, 

**Raspberry Pi Zero W - MotionEyeOS**

1. Download the [latest img](https://1drv.ms/u/s!AmNBhv72BI68got82f3cirzYG0bfRw) for MotionEyeOS -  recompiled from the updated firmware
1. Install in the [usual ways](https://github.com/ccrisan/motioneyeos/wiki/Installation)
1. Once flashed, open the boot partition and add a file called wpa_supplicant.conf 
1. Open and edit wpa_supplicant.conf with the instructions from [here](https://github.com/ccrisan/motioneyeos/wiki/Troubleshooting-Raspberry-PI-Models-A-(Plus)-And-Zero)
1. Once booted, it should connect to the wireless automatically and you can just find the IP and log in

**Notes:** 
- All of this has been compiled from the repository, no changes on my part
- This is not officially supported yet, so don't annoy @ccrisan if this doesn't work
- I've tested it with 2 Pi Zero W's with the official camera module - I haven't tested it with any usb cameras
- No drivers need to be copies or any files modified

30dbs, 2017-03-07:
great stuff @wdawson001 it works perfectly.

k-e-y-s-a, 2017-03-07:
yep, bonzer, thanks

Gantelope, 2017-03-07:
Thanks, worked first time!

Gantelope, 2017-03-08:
Hi, trying to log in via SSH fails with this build. Can anyone else connect or is it just me?

wdawson001, 2017-03-08:
Have you enabled it in the webpage? 

SSH is not enabled by default in raspbian any more. To enable it. Put a blank file called SSH (no extention) on the boot partition.

Gantelope, 2017-03-09:
Is this applicable to MotionEyesOS? I have enabled SSH in the administrator account settings. Do I need to do more than that?

jezmck, 2017-03-10:
Thanks @wdawson001! Works perfectly.

wdawson001, 2017-03-10:
@Gantelope - I've just tried SSH on this build and I can't get it to work. This is not a official build, so it may be something that will get addressed in the next release.

Gantelope, 2017-03-10:
Thanks for checking. Good to know it's not something I have done.
As a newbie I hadn't realised Google Drive could be configured from the GUI so I was trying to use SSH to do this. Google Drive is up and running now.

I have to say I am so impressed with all of this. It works so well. I have had it running for a few days now without any problems. Even spotted my window cleaner working hard this morning!

Thanks for all the effort put into MotionEyesOS.

derrickmehaffy, 2017-03-13:
I was able to get the system up and running and during the process I created the empty file SSH however the service fails to start because host keys were no generated. I tried using ssh-keygen but it wasn't outputting the files in /data/etc/ anyone else having an issue like this?

wagsforever, 2017-03-14:
Hi guys. Sorry if i'm being a noob but I have the Zero W and trying to get this running and obviously it says no camera as others have found. The fix above is for the motioneyeos. Is there anyone to get it working on motioneye on raspbian??

Zitman, 2017-03-21:
Thanks all,

Using the info and image provided by @wdawson001 I got mine up and running, connected to my office wireless network and camera.  However when I amend the details in wpa_supplicant.confto the settings for my home network it will not connect.  I have got MotionEye running on another Pi Zero (not W) perfectly and have copied the wpa_supplicant.conf from this one to the ZeroW and still no good. 

If I plug a monitor in it just sits at the connecting to WIFI bit then says no wireless connection.

Not sure what I might be missing, any thoughts?

Z

Jay-Arr, 2017-03-27:
The new official March build works correctly.  If it still doesn't work for you, read this:

https://github.com/ccrisan/motioneye/issues/497#issuecomment-289584141

pshotton, 2017-04-12:
I can't find the official march build. Latest on the install page of the wiki is 20170212.

I tried the build supplied with wdawson001 and wifi works fine, but I still can't get the official 5Mb Zero camera to work. If I edit config.txt and add the 'start_x=1' line it refuses to boot :(

Guess I'll have to wait - I appreciate the massive amount of work ccrisan puts in on this and I'll be patient :)

pshotton, 2017-04-12:
Realised I needed the fixup_x.dat and start_x.elf files as well (should read the post better). Now it boots but wifi has stopped working :(

ccrisan, 2017-04-12:
@pshotton https://github.com/ccrisan/motioneyeos/releases/tag/20170329

pshotton, 2017-04-12:
Superb. That's fixed the sshd problem. However local camera still not working. Hope I haven't broken it pulling it in and out several times :(

Anyway - superb job to get a pre-release out. Beer tokens on their way.

ccrisan, 2017-04-13:
@pshotton thanks. Try using Raspbian and see if you can make the camera work with it.

diarem, 2017-04-20:
Hi, my application is slightly different to the thread but I am hoping someone will know straight away what I am doing wrong!  I am using a Raspberry Pi B with a USB video grabber connected to a bird box camera. The software loaded first time, I can access the Pi easily and configure everything - except that it can't find a local camera! I am not trying to use wireless, I am happy with an Ethernet cable but I have tried the various suggestions about adding files without success.
Can anyone point me in the right direction please?

cl0udb0y, 2017-04-27:
Hi is there any progress I have missed in other posts? same issues with WiFi and Local Camera.

I have a couple of piZero W. OTB observation - latest MotionEye OS halts on Ethernet - added the wpa_supplicant.conf doesn't work for me - if I boot via USB eth then fine - enabling wifi in MotionEye UI and rebooting trys to start wpa but can't find device. 

I have an older piNoir and new and neither can be found with motioneyeOS

I have tried latest raspbian image and wifi works fine as does the camera
Installing motioneye works inc wifi - the camera only works after adding the module and a hard reboot.

I'd rather use the optiomised MotioneyeOS than raspian + build - happy to test / post results if it helps.

John

prabhuvd, 2017-05-21:
I'm using latest motion eye image on pi3 , still have issues with wifi access , I have setup the wpa_supplicant.conf file in /data/etc/wpa_supplicant.conf but there is no /var/run/wpa_supplicant file.
Hence during bootup I get error: starting wpa_supplicant:failed (brcmfmac )

 Has anyone faced similar issues?

bortek, 2017-06-01:
Just tested the latest image on pi zero W and NOIR camera, a diffrent NOIR camera, pi1 Camera, PI2 camera all works fine!

This is not Motioneyeos issue. Check your hardware and test your setup on regular raspbian!

closed 2017-06-01 by bortek

prabhuvd, 2017-06-01:
Found the problem  !!  Motioneye did not like  

> key_mgmt="WPA-PSK"

 in  wpa_supplicant.conf . 
The problem was fixed after i deleted the above line

JoeRelic, 2018-01-12:
Using MotioneyeOS (20180101) but won't recognize the Pi NoIR camera. Wireless working well.