motioneye-project/motioneyeos #811: No local cameras detected on Pi Zero 1.3 with Pi Camera

rwkiii, 2017-03-05:
I had a difficult time getting a Pi Zero setup due to the networking issue, but was able to get it working once I put the wpa_supplicant.conf file on the SD card before booting.

Once the networking issue was out of the way I installed the camera, but it didn't get detected by the MotionOS web UI. Then I wondered if the camera had to be installed on first bootup also.

I reflashed the SD with the Raspberry PI image, copied the wpa_supplicant.conf to the SD, installed the camera. Rebooted the Pi Zero.

Still, no local cameras. The camera is a 5mp official Pi Camera connected using the Pi Zero's onboard camera connector.

**motioneye.log:**

```
2017-03-05 18:38:02: [motioneye]     INFO: hello! this is motionEye server 20170212
2017-03-05 18:38:03: [motioneye]     INFO: main config file /data/etc/motion.conf does not exist, using default values
2017-03-05 18:38:04: [motioneye]     INFO: cleanup started
2017-03-05 18:38:04: [motioneye]     INFO: wsswitch started
2017-03-05 18:38:04: [motioneye]     INFO: tasks started
2017-03-05 18:38:04: [motioneye]     INFO: mjpg client garbage collector started
2017-03-05 18:38:04: [motioneye]     INFO: smb mounts started
2017-03-05 18:38:04: [motioneye]     INFO: server started
2017-03-05 18:45:48: [motioneye]    ERROR: [Errno 2] No such file or directory: '/var/log/motion.log'
Traceback (most recent call last):
  File "home/ccrisan/motioneyeos/output/raspberrypi/target/usr/lib/python2.7/site-packages/tornado/web.py", line 1443, in _execute
  File "/usr/lib/python2.7/site-packages/motioneye/handlers.py", line 175, in wrapper
  File "/usr/lib/python2.7/site-packages/motioneye/handlers.py", line 1750, in get
```

**messages.log:**

```
Jan  1 00:00:06 meye-8f3c5823 auth.err passwd: /etc/passwd: Read-only file system
Jan  1 00:00:06 meye-8f3c5823 auth.err passwd: can't update password file /etc/passwd
Jan  1 00:00:06 meye-8f3c5823 auth.err passwd: /etc/passwd: Read-only file system
Jan  1 00:00:06 meye-8f3c5823 auth.err passwd: can't update password file /etc/passwd
Jan  1 00:00:19 meye-8f3c5823 daemon.info dhclient: DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 3
Jan  1 00:00:19 meye-8f3c5823 daemon.info dhclient: DHCPREQUEST on wlan0 to 255.255.255.255 port 67
Jan  1 00:00:19 meye-8f3c5823 daemon.info dhclient: DHCPOFFER from 10.10.10.1
Jan  1 00:00:19 meye-8f3c5823 daemon.info dhclient: DHCPACK from 10.10.10.1
Jan  1 00:00:20 meye-8f3c5823 daemon.info dhclient: bound to 10.10.10.58 -- renewal in 252 seconds.
Mar  5 18:37:55 meye-8f3c5823 cron.info crond[826]: crond (busybox 1.24.2) started, log level 8
Mar  5 18:37:55 meye-8f3c5823 cron.info crond[829]: crond (busybox 1.24.2) started, log level 8
Mar  5 18:37:55 meye-8f3c5823 auth.info sshd[845]: Server listening on 0.0.0.0 port 22.
Mar  5 18:37:55 meye-8f3c5823 daemon.notice proftpd[860]: localhost - ProFTPD 1.3.5b (maint) (built Sun Feb 12 2017 12:16:05 EET) standalone mode STARTUP
Mar  5 18:38:13 meye-8f3c5823 daemon.info init: starting pid 1046, tty '/dev/ttylogin': '/sbin/getty -L ttylogin 115200 vt100'
Mar  5 18:42:04 meye-8f3c5823 daemon.info dhclient: DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 3
Mar  5 18:42:04 meye-8f3c5823 user.notice network: wlan0 has no IP address
Mar  5 18:42:05 meye-8f3c5823 daemon.info dhclient: DHCPREQUEST on wlan0 to 255.255.255.255 port 67
Mar  5 18:42:05 meye-8f3c5823 daemon.info dhclient: DHCPOFFER from 10.10.10.1
Mar  5 18:42:05 meye-8f3c5823 daemon.info dhclient: DHCPACK from 10.10.10.1
Mar  5 18:42:05 meye-8f3c5823 daemon.info dhclient: bound to 10.10.10.58 -- renewal in 136 seconds.
Mar  5 18:44:21 meye-8f3c5823 daemon.info dhclient: DHCPREQUEST on wlan0 to 10.10.10.1 port 67
Mar  5 18:44:21 meye-8f3c5823 daemon.info dhclient: DHCPACK from 10.10.10.1
Mar  5 18:44:21 meye-8f3c5823 daemon.info dhclient: bound to 10.10.10.58 -- renewal in 292 seconds.
```

**boot.log:**

```
---- booting motionEyeOS 20170212----
 * Mounting filesystems: done
 * Loading kernel modules: done
 * Setting hostname: done
 * Starting syslogd: done
 * Configuring CPU frequency: done
 * Starting watchdog: done
 * Starting brownout detector: done
 * Starting eudev: done
 * Setting root password: done
 * Starting wpa_supplicant: done (8192cu)
 * Configuring wireless network: dhcp
 * Setting current date using http: Sun Mar 5 18:37:55 UTC 2017
[... 231 lines cut ...]
[Thu Jan  1 00:00:17 1970] WARNING: CPU: 0 PID: 6 at net/wireless/sme.c:714 __cfg80211_connect_result+0x2a8/0x3d8 [cfg80211]()
[Thu Jan  1 00:00:17 1970] Modules linked in: 8192cu cfg80211 rfkill snd_soc_bcm2835_i2s regmap_mmio snd_bcm2835 snd_soc_core snd_pcm_dmaengine snd_pcm snd_timer snd i2c_bcm2708 spi_bcm2835 bcm2835_gpiomem uio_pdrv_genirq uio bcm2835_wdt bcm2835_v4l2 videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_core v4l2_common videodev media
[Thu Jan  1 00:00:17 1970] CPU: 0 PID: 6 Comm: kworker/u2:0 Not tainted 4.4.38 #1
[Thu Jan  1 00:00:17 1970] Hardware name: BCM2708
[Thu Jan  1 00:00:17 1970] Workqueue: cfg80211 cfg80211_event_work [cfg80211]
[Thu Jan  1 00:00:17 1970] [<c0016cfc>] (unwind_backtrace) from [<c0013c20>] (show_stack+0x20/0x24)
[Thu Jan  1 00:00:17 1970] [<c0013c20>] (show_stack) from [<c02e437c>] (dump_stack+0x20/0x28)
[Thu Jan  1 00:00:17 1970] [<c02e437c>] (dump_stack) from [<c0021ef4>] (warn_slowpath_common+0x8c/0xc4)
[Thu Jan  1 00:00:17 1970] [<c0021ef4>] (warn_slowpath_common) from [<c0021fe8>] (warn_slowpath_null+0x2c/0x34)
[Thu Jan  1 00:00:17 1970] [<c0021fe8>] (warn_slowpath_null) from [<bf161568>] (__cfg80211_connect_result+0x2a8/0x3d8 [cfg80211])
[Thu Jan  1 00:00:17 1970] [<bf161568>] (__cfg80211_connect_result [cfg80211]) from [<bf13d088>] (cfg80211_process_wdev_events+0x17c/0x20c [cfg80211])
[Thu Jan  1 00:00:17 1970] [<bf13d088>] (cfg80211_process_wdev_events [cfg80211]) from [<bf13d154>] (cfg80211_process_rdev_events+0x3c/0x74 [cfg80211])
[Thu Jan  1 00:00:17 1970] [<bf13d154>] (cfg80211_process_rdev_events [cfg80211]) from [<bf138208>] (cfg80211_event_work+0x24/0x2c [cfg80211])
[Thu Jan  1 00:00:17 1970] [<bf138208>] (cfg80211_event_work [cfg80211]) from [<c00376a4>] (process_one_work+0x11c/0x39c)
[Thu Jan  1 00:00:17 1970] [<c00376a4>] (process_one_work) from [<c0037964>] (worker_thread+0x40/0x4d0)
[Thu Jan  1 00:00:17 1970] [<c0037964>] (worker_thread) from [<c003d2e8>] (kthread+0xdc/0xf8)
[Thu Jan  1 00:00:17 1970] [<c003d2e8>] (kthread) from [<c000f8a8>] (ret_from_fork+0x14/0x2c)
[Thu Jan  1 00:00:17 1970] ---[ end trace d1ade9012ca5f967 ]---
[Thu Jan  1 00:00:18 1970] IPv6: Loaded, but administratively disabled, reboot required to enable
[Thu Jan  1 00:00:20 1970] system date is 2017-03-05 18:37:55
```

There is no motion.log.

Does anyone see a clue?

ccrisan, 2017-03-05:
It looks like your OS did not successfully finish its first boot. You might have interrupted it when you were playing with it at the beginning, or something else could have gone wrong. I recommend rewriting the entire OS from scratch, this time making sure to have the correct `wpa_supplicant.conf` on the boot partition, before the first.boot.

rwkiii, 2017-03-05:
I wrote the .img to SD, copied the wpa_supplicant.conf to the SD. Ejected it from my PC (Windows 10) and inserted it into the Pi Zero. Then plugged the Pi Zero into power and left it alone for 5 minutes. It came up on the wireless network fine. I accessed it via the web UI and still, no local camera.

I installed the camera with the metal contacts facing the Pi's PCB which I think I can see is the contact side.

ccrisan, 2017-03-05:
Does it work with Raspbian?

rwkiii, 2017-03-05:
I have an SD card I can put in it to test. I'll give it a try. Is it good enough to just use raspistill for the test?

ccrisan, 2017-03-05:
I think raspistill should be enough.

rwkiii, 2017-03-05:
It isn't working in Raspian either. There isn't any obvious issues. It's all new, connected correctly, plenty of power. I'll update the thread with what I find.

ccrisan, 2017-03-05:
I'll close the issue since it's not about motionEyeOS. But please, keep us posted :)

closed 2017-03-05 by ccrisan