motioneye-project/motioneyeos #812: Setting GPIO pin state remotely from server

MaartenMens, 2017-03-06:
Hello,

I'll first describe my setup.
I have installed MotionEye on an Odroid xu4 running Ubuntu Mate 15.04. 
I have a RPi3 with raspi cam NoIR v2 running MotionEyeOs which is set up as fast network camera. 
I added the RPi3 fast network camera to MotionEye as network camera. 
The Odroid and RPi3 are both on the same local network. 

Is it possible to have an action button on the MotionEye server set the state of a GPIO pin on the RPi3 fast network camera? I know how to enable the action button in motioneye on the Odroid, but I don't know what code to use to access the GPIO pins on the RPi3. If this can be done, could it be done with a simple python script or is it more elaborate? 

I was not sure whether to put this issue in the motioneye or in the motioneyeos section. 

Hope someone could give me an example on how to go about this.

ccrisan, 2017-03-06:
Normally this can be done via SSH. Just make sure to add the id_rsa.pub (or similar) key of your OdroidXU4 to the `/data/etc/ssh_authorized_keys` of your Raspberry PI.

ccrisan, 2017-03-06:
See [this](https://github.com/ccrisan/motioneyeos/wiki/Action-Buttons) for the commands to be run on your RPi to control the GPIOs via a shell script. You can also find this info online, as it has nothing to do with motionEyeOS.

closed 2017-03-06 by ccrisan