motioneye-project/motioneyeos #814: Web UI can't be opened from 3G

toothGandalf, 2017-03-07:
I have this issue for almost year, with different motioneyeos versions. Each time when i'm trying to connect to web interface from 3G internet i'm getting this message "An error occured. Refreshing is recommended". If my phone is connected to any kind of WiFi network everything works fine. After successful login i can use web ui on any network while browser tab not closed. I've tried clearing cache, opening in incognito mode even in different browsers, results are the same.

ccrisan, 2017-03-07:
First off, motionEyeOS does not support 3G modems. There's some kind of preliminary work in this direction, but has never been tested.

Secondly, I'd like to see your motioneye.log file after such an error occurs.

toothGandalf, 2017-03-07:
Hi, thanks for your reply and sorry for my English. I was trying to say that this issue occurs when my phone is on 3G network and is trying to access web ui from my phone. When phone is connected to wifi everything works fine.
This is what i see when i'm trying to connect trough 3G
[image]

This is my recent events in motioneye.log (46.211.155.205 and 46.211.155.199 is my mobile phone on 3G):

ccrisan, 2017-03-07:
When your phone is connected to the mobile network, it can't also be connected to your wifi network. If you want your motionEyeOS to be accessible from the internet, in general, you'll need to set up port forwarding on your router. But that's beyond the scope of motionEyeOS.

closed 2017-03-07 by ccrisan

toothGandalf, 2017-03-07:
I've set my router port forwarding and i can connect to motioneye from any other place trough mobile browser but **only** when internet connection type is wifi

ccrisan, 2017-03-07:
I'm sorry but your log file does not show any error that could be related to what you get on your mobile phone screen. I have no idea what could cause that problem.

toothGandalf, 2017-03-07:
Okay, i've managed to figure out what is happening.
If it matters.
It's caused by custom 403 error responce from my cellular network provider.
Instead of getting plane json with error message, motioneye web ui is getting this page.
Because my provider decided that internet needs custom error pages :(
[image]

ccrisan, 2017-03-07:
Thanks for keeping us posted. I'm really sorry about your provider but there's nothing I can do about it.