motioneye-project/motioneyeos #819: Raspberry 2 is stuck onboot

visiongeist, 2017-03-11:
Hi,

I just installed the latest ("motioneyeos-raspberrypi2-20170212.img") on my raspberry but i cannot get it to boot beyond the rainbow screen (red led on). Since I do not have any logs, I was wonderin is there is any recommendation how to fix this. I already tried 2 different SDs in case of a card faulty.

Regards,
Damien

ccrisan, 2017-03-11:
See https://github.com/ccrisan/motioneyeos/wiki/Troubleshooting#system-wont-boot.

visiongeist, 2017-03-11:
@ccrisan i verified it again. Noobs is working with the same SD card. :-/ I tried several tools to setup the SD card, but no success so far

ccrisan, 2017-03-11:
What is your exact board model, as written on the PCB?

visiongeist, 2017-03-11:
rasp pi model b+ v1.2

ccrisan, 2017-03-11:
I believe that speaks for itself :)

closed 2017-03-11 by ccrisan

visiongeist, 2017-03-11:
ah damn it :( ok sorry for the noise. i ll get new one

visiongeist, 2017-03-11:
now it works... stupid me ;)