motioneye-project/motioneyeos #821: Initial startup raspberrypi 3

pgas37, 2017-03-12:
I have raspberrypi 3 model b
 I wrote the motionpi –Rapsberrypi2version 20150719 image (latest) on the sd-card ( 32 gb) with win32diskmanager. Succesfull write.
 
According the wiki- documentation the initial startup take 1 or 2 minutes. I waited at least 10 minutes but nothing happens. No ethernet lights and/or pi light is blinking. The pi seem completely dead.
I have tried this everal times.
The same result with the raspberrypi-version20150719.

With a raspbian-image the pi and camera are working well.

Mr Crisan advised me to write the issue here.

Any help will be appriciated.

Paul

wdawson001, 2017-03-12:
Hi @pgas37,

It looks like you are writing the wrong image - both version and wrong board.

You need to download the latest raspberrypi3 img from [here](https://github.com/ccrisan/motioneyeos/releases/tag/20170212)

Try again and let me know how you get on. Also, are you able to attach a monitor?

pgas37, 2017-03-12:
I followed the link https://github.com/ccrisan/motionPie/releases. That link does not show the pi3.img!

the pis3 img installed:
On 192.xxx meye is visible ,but with very limited preferences. Only Fit frame vertically, layout rows, frame rate dimmer and resolution dimmer. So nothing els!
a monitor can be attached, but i have to login, but what is the login and ww?

regards

ccrisan, 2017-03-12:
@pgas37 Logout and log in as administrator. Then make sure the advanced settings option is on.

closed 2017-03-13 by pgas37