motioneye-project/motioneyeos #823: i2c not fully available, no i2cdetect

cuteboi, 2017-03-13:
I'm on RPI3, and unable to load up the pcf8523 from adafruit, will go to them if this isn't a motioneyeos issue. 
motioneyeos-raspberrypi3-20170212.img

```
# cat /data/etc/modules 
rtc-pcf8523
```

```
# lsmod | grep rtc
rtc_pcf8523             3003  0 
```

```
# i2cdetect -y 1
i2cdetect: can't open '/dev/i2c/1': No such file or directory
```

```
# cat /data/etc/userinit.sh 
echo pcf8523 0x68 > /sys/class/i2c-adapter/i2c-1/new_device
hwclock -s
```

Tested the commands for userinit.sh:

```
# echo pcf8523 0x68 > /sys/class/i2c-adapter/i2c-1/new_device
-sh: echo: write error: Invalid argument
```

and of course hwclock wont work without the i2c binds in place.

```
# hwclock -s
hwclock: can't open '/dev/misc/rtc': No such file or directory
```

Followed instructions from the tweaks page.

ccrisan, 2017-03-13:
I don't like the following message:

    `i2cdetect: can't open '/dev/i2c/1': No such file or directory`

Someone with more experience could enlighten us?

cuteboi, 2017-03-15:
I tested with the latest Raspbian, it worked flawlessly.  I'll come back to this project a little later for my car's side view camera setup later.  This device needs to work in an offline setup, waiting for GPS is going to delay things, and using the RTC would speed things up.

ccrisan, 2017-03-15:
@cuteboi can you try previous versions of motionEyeOS? I remember users managing to have RTC with motionEyeOS. Something might have gotten broken with recent versions with regards to the I2C (some DTB missing?)

lapsedtheorist, 2017-04-17:
@ccrisan In case it helps, I've been able to get i2c in motioneyeos 20170212 working on an RPi A+ by adding `i2c-dev` to `/etc/modules` (or `/data/etc/modules`). It seems this is the necessary module to allow `i2cdetect` to work.

ccrisan, 2017-04-17:
@lapsedtheorist does this also work with 20170329?

lapsedtheorist, 2017-04-17:
@ccrisan Sorry, I haven't had the opportunity to test that release yet as I'm in the middle of a project and can't flatten the card I'm working from.

From experience of i2c on Raspbian, the dependencies appear to be that `/boot/config.txt` contains `dtparam=i2c_arm=on` and that the module `i2c-bcm2708` has been loaded. If those are satisfied (as they appear to be), I can't see why simply adding `i2c-dev` to the modules list shouldn't be enough.

lapsedtheorist, 2017-05-06:
@ccrisan I'm happy to confirm that adding `i2c-dev` to `/etc/modules` on a fresh install of `20170329` allows `i2cdetect` and friends to work.

ccrisan, 2017-05-07:
f3f1b6c2922bd27267961892d1a323162f06844e fixes this.

closed 2017-05-07 by ccrisan